Atmosphere-NX/Atmosphere #2496: Omelet superdeck doesn't work under atmosphere

veenone, 2025-03-16:
### What's the issue you encountered?

I'm not sure if this really related to atmosphere at all, but I just bought omelet superdeck :

https://omelet.store/collections/deck-controller-for-nintendo-switch-nintendo-switch-oled?srsltid=AfmBOoocAcuIO-HD8MHG8NYLH2nHcKrhuVEgleHAzqoF4FrkbZE-3RvX
 The problem is this accessory only work with stock firmware. When I restart the fw to atmosphere, the device doesn't work at all. 

To enable this, the only configuration is to enable wired switch pro controller setting.

Currently I'm using version 1.8.0

### How can the issue be reproduced?

[ * ] Just start switch under atmosphere with connected accessory 

### System Firmware Version

1.9.0.1

### Environment?

- Hekate
- Official release or unofficial build: 
  - [ Official release version 6.2.2 ]

SciresM, 2025-03-16:
I don't really know of any mechanism by which atmosphere could be interfering with this. Can you post a screenshot of your atmosphere/contents folder, and also your system settings ini, and also your Hekate config, so I can evaluate if this is really a base atmosphere problem?

veenone, 2025-03-17:
Thanks for your reply. I will share these information later on today after work. Additionally, I use kefir to install the firmware. And the latest version one. Not sure if this additional information helps.

Masamune3210, 2025-03-17:
Backup sd card to something else
Wipe it
Install Atmosphere the correct way with no sysmodules or patches
Recreate Issue

Masamune3210, 2025-03-17:
Until then, its unlikely that any more support will be given, considering Keifr and especially their version of atmosphere are considered piracy tools

Xplosios, 2025-03-20:
hello i ran into the same problem have you found the solution?