Atmosphere-NX/Atmosphere #2497: A fatal error occured when running Atmosphere. Program ID: 524850534f53454d

JSketch, 2025-04-03:
## Bug Report

Error Desc : Kernel Panic (0xf00)

### What's the issue you encountered?

Today I installed the hwfly correctly, I tried to install atmosphere many times but when I run Atmosphere appear this issue (when I run the stock boot all is normal)

Before that the fusee error appear and I downloaded the fussed on GitHub and paste that in payload folder. Now I got this problem.

### How can the issue be reproduced?

Idk

### Crash Report

### System Firmware Version

Atmosphere 1.8.0 

### Environment?

Atmosphere boot

### Additional context?

Nop:)

SciresM, 2025-04-03:
You forgot to include an error report file.

JSketch, 2025-04-03:
I downloaded atmosphere again and copied it to the SD, the problem is solved.

closed 2025-04-03 by JSketch