Atmosphere-NX/Atmosphere #2500: CFW related faulty button mapping while using Taiko Drum Controller

Daemsak, 2025-04-23:
## Bug Report

### What's the issue you encountered?

I am running latest Atmosphere 1.8.0. 
Issue is related to a specific controller.
Taiko Drum controller for switch, the button mapping is faulty when used with CFW, when i load into Stock CFW everything works as intended.(have checked 3 other switch Docks and switches, just to be sure)

### How can the issue be reproduced?

Connect a Taiko drum pad to a Switch running Atmosphere and check the a/b/x/y button mapping
(its also wrong on the drum itself which makes it unusable)

### Crash Report

N/A

### System Firmware Version

19.0.1

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee
- Official release or unofficial build: Official release, imported via Kefir
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND: SysNAND

### Additional context?

- Additional info about your environment: No

Daemsak, 2025-04-24:
Just realzied was using a modded Atmosphere, problem is the same on the stock version tho

borntohonk, 2025-04-24:
> Official release or unofficial build: Official release, imported via Kefir

piracy isn't supported.

anyways, boot without atmosphere, go to eshop, download the demo (free) and test if the controller is wrong there.

if it is wrong there, then your product is wrong, not cfw related.

Daemsak, 2025-04-24:
@borntohonk 

> anyways, boot without atmosphere, go to eshop, download the demo (free) and test if the controller is wrong there.

Have tried that, everything works as intended on stock FW(im referring to Horizon OS here).

Does not work fine(buttons are messed up and some dont work) with both the "pirated" and official versions of Atmosphere.

Sorry if anything i've said is missleading. I dont have a lot of conversations regarding this topic, moreover in english.

I am also aware that this is a very minor issue, just decided it might be useful for the cfw devs/other users to know about it.

SciresM, 2025-04-24:
Can you post a screenshot of your atmosphere/contents folder?

Daemsak, 2025-04-25:
Thank you to everyone who payed attention to this issue, i appreciate the community. 
Especially @SciresM, who is doing a great job as an Atmosphere dev and manages to keep up with the issues community submits.

Daemsak, 2025-04-25:
This issue has been related to Sys-con module which took over the usb interface for the compatablity purposes.

When testing i did not notice it was installed on both test CFW switches.

If you are having issues with Taiko drum pad and a custom firmware, look for additional modules installed on your system, this might be the cause of your issue.

closed 2025-04-25 by Daemsak