Atmosphere-NX/Atmosphere #2502: 20.0.0 Support

SciresM, 2025-04-30:
Master issue/check list of stuff to do for 20.0.0 changes. Noting that work on this is being [streamed on Twitch](https://www.twitch.tv/sciresm). 

- [x] Reverse engineer new kernel changes 
- [x] Update fusee/exosphere with new keydata/values
- [ ] Implement new changes into mesosphere
- [ ] Implement new changes into erpt
- [ ] Implement new changes into ncm
- [ ] Implement possible new changes into ns mitm (if needed)
- [ ] Implement new changes into <sysmodules, create as needed>
- [x] Update emummc
- [ ] Miscellaneous exosphere/fusee fixes for new DRAM IDs
- [ ] Make sure libnx is updated for anything relevant.

Post-release:
- [ ] Determine what other system modules need reverse engineering changes
- [ ] Reverse engineer potential new boot sysmodule changes.
- [ ] Miscellaneous post-release reverse engineering
- [ ] Miscellaneous post-release implementation work/updating

SciresM, 2025-04-30:
Work is being done on the [20.0.0 support branch](https://github.com/Atmosphere-NX/Atmosphere/tree/20_support).

borntohonk, 2025-04-30:
I left comments on: https://github.com/Atmosphere-NX/Atmosphere/commit/bd0890affba591ec791f841c148c2931d2b800f2
as the key vectors for at least production and development doesn't seem to be encrypted correctly (i cannot check the "device", but i assume those also probably are wrong if those two are wrong)

pattyfusty, 2025-04-30:
How long will this take?

rashevskyv, 2025-04-30:
> How long will this take?

As long as it takes

SciresM, 2025-04-30:
@borntohonk it says you left comments, which look deleted, may I assume your commentary on key vector correctness was bogus?

fwiw (as you may know from my livestream), the keydata vectors are generated using a script, so it would surprise me a lot if they were wrong.

ELCALLEJONGAMER, 2025-04-30:
Good morning, is there any way to generate the blown fuses warmboot_mariko so I can start the emummc with sleep?

> Work is being done on the [20.0.0 support branch](https://github.com/Atmosphere-NX/Atmosphere/tree/20_support).

borntohonk, 2025-04-30:
> [@borntohonk](https://github.com/borntohonk) it says you left comments, which look deleted, may I assume your commentary on key vector correctness was bogus?
>
> fwiw (as you may know from my [...]

No I noticed an error in *my own* script, which i accidentally encrypted master_key_00 instead of master_key_12.

XarceChronus, 2025-04-30:
This update is pretty substantial so I imagine it will take awhile. 
Especially with the virtual game cards possibly affecting dumps. 
But what do I know? (Nothing actually)

borntohonk, 2025-04-30:
> This update is pretty substantial so I imagine it will take awhile. 
> Especially with the virtual game cards possibly affecting dumps. 
> But what do I know? (Nothing actually) 

This comment is extremely inappropriate, piracy use case is not supported.

SciresM, 2025-04-30:
This issue is for tracking my work. Please keep petty squabbles out of my issue. I will ban anyone who posts anything inane on this issue (most of the above was) for a brief period of time.

SonicStylin91, 2025-04-30:
gotcha

SonicStylin91, 2025-04-30:
how long will this take?

DamianTheDude, 2025-05-01:
I wait for this with mild impatience but not anger

kirakira-commits, 2025-05-02:
Very happy to see Atmosphere being updated so quick after the new firmware!
Am looking very forward to this

vanmoniq, 2025-05-02:
💪🙏

Soneoylys, 2025-05-02:
> I wait for this with mild impatience but not anger

Check his stream at night in PT, I actually enjoy to see how it progressing and challenges we saw and various interesting solutions come up with.
We have a great time in stream everyday, like tonight we saw the possibly biggest problem for upgrade (big N alloc more memory than before and added freem checks), so why not you join us together, maybe you will understand the difficulty and hopefully you will find your patience after.

nriver007, 2025-05-02:
Is this going to also support 20.0.1?

neio4, 2025-05-02:
> Is this going to also support 20.0.1?
 
#2505

3pochs, 2025-05-02:
> Check his stream at night in PT, I actually enjoy to see how it progressing and challenges we saw and various interesting solutions come up with. We have a great time in stream everyday, like [...]

I abesloutely love the streams I'm planning to come every night it's awesome!

Ch3ck3rM0n, 2025-05-02:
> 
> I abesloutely love the streams I'm planning to come every night it's awesome!

Its the same to me. So I can support a little bit.

souldbminersmwc, 2025-05-02:
Did 20.0.1 change anything with the progress of 20.0.0 support?

SciresM, 2025-05-02:
No.

OM3GAZX, 2025-05-03:
Eagerly awaiting for the next update. Keep it up, SciresM!
We're counting on you. Best of luck!

costelabrfan, 2025-05-03:
Wee need u ...

t3422800lt, 2025-05-03:
I mean, will the new 20.0.0 atmosphere and the fusee.bin file that you are currently working on work with the latest Nintendo switch system update version 20.0.1?

t3422800lt, 2025-05-03:
Ok. Thank you for letting me know, but would you (Soneoylys) mind asking SciresM himself if what you (Soneoylys) said is true so I can get a better confirmation?

Soneoylys, 2025-05-03:
Only one jpeg loader gets updated, you can check full changelog at [here](https://switchbrew.org/wiki/20.0.1)

9Bitdo, 2025-05-03:
What scires is doing for 20.0.0 will support 20.0.1 as well. Just seek everyone understanding to be patient and wait work to be done. 

No need to keep asking when atmosphere will be release, a stable version will come when its ready

SciresM, 2025-05-03:
An initial QA testing build was posted on the ReSwitched discord and in my stream chat. 20.0.1 works totally fine.

Once a day passes without major issues reported, I'll upload a build to this issue's comments for further testing, and a few days after that it'll become a pre-release.

Thank you for your continued patience.

winmetawin0, 2025-05-03:
언제쯤 닌텐도를 할 수 있나요

impeeza, 2025-05-03:
> 언제쯤 닌텐도를 할 수 있나요

Be polite. People are making the changes on their own time and without any compensation. And people demanding on harsh way is not nice.

winmetawin0, 2025-05-03:
> 언제쯤이면 할 수 있나요?

⭐밥먹고 올테니 빠른처리^^⭐

winmetawin0, 2025-05-03:
표현의 자유😆

9Bitdo, 2025-05-03:
표현의 자유와 무례함은 전혀 다른 문제다

Nazgile94, 2025-05-03:
Take all time needed and Big thanks for working in 20.0.0 ❤️ 

Heros !

winmetawin0, 2025-05-03:
8282828282

shwissjeez, 2025-05-03:
THANK YOU FOR EVERYTHING YOU DO ❤️❤️❤️❤️❤️❤️❤️

sir1st, 2025-05-04:
is it possible to add a docker env for ams make? i'd like to add this file but requires some information, could you offer some help?

hanbalfrek, 2025-05-04:
Sorry for being a total beginner at this, and I know this isnt a support page, but I've been troubleshooting for about 2 hours now.  About a month ago I got a modded switch working for my son using Atmosphere and Hekate and Arcropolis using sysMMC.  However, this morning my son clicked update on something as he was starting Smash, and now when launching from Hekate, it says Unknown pkg1 version.  HOS version is not supported.
I figured he updated the modded copy in the SD Card, and so I:
- copied in my SD Card backup, same problem
- deleted everything, unzipped fresh downloads (same versions) of Atmosphere and Hekate, same problem
- formatted the SD card again, copied in fresh copies of everything, same problem.

Now I see you're working on version 20.  Is it that he updated the Switch firmware, and I just now have to wait for your update?
Or is there something simpler I'm missing?

Continous, 2025-05-04:
> Sorry for being a total beginner at this, and I know this isnt a support page, but I've been troubleshooting for about 2 hours now. About a month ago I got a modded switch working for my son using Atmosphere and Hekate and Arcropolis using sysMMC. However, this morning my son clicked update on something as he was starting Smash, and now when launching from Hekate, it says Unknown pkg1 version. HOS version is not supported. I figured he updated the modded copy in the SD Card, and so I:
> 
>     * copied in my SD Card backup, same problem
> 
>     * deleted everything, unzipped fresh downloads (same versions) of Atmosphere and Hekate, same problem
> 
>     * formatted the SD card again, copied in fresh copies of everything, same problem.
> 
> 
> Now I see you're working on version 20. Is it that he updated the Switch firmware, and I just now have to wait for your update? Or is there something simpler I'm missing?
> 
> Much thanks in advanced!

Yes, likely, your son updated to 20.x which is not yet supported by public releases. Wait for the next release or boot into stock HOS.

SciresM, 2025-05-04:
fwiw, current status is investigating a known issue where some consoles hang forever on the Switch logo, booting into 20.0.0.

WuBoytH, 2025-05-04:
> fwiw, current status is investigating a known issue where some consoles hang forever on the Switch logo, booting into 20.0.0.

If it helps, I noticed this happening with my V1 Switch. It stopped happening after I let it boot stock once, though I understand people won't want to do that to avoid burning fuses.

SciresM, 2025-05-04:
There are reports of it not being fixed by booting stock successfully.

SciresM, 2025-05-04:
Okay -- I am still going to give this a few extra days to simmer before doing a proper pre-release.

However, if you wish to use atmosphere before it is fully ready, here's a build with which you can do so.

[atmosphere-1.9.0-preprerelease-20_support-9dd8269f7+hbl-2.4.4+hbmenu-3.6.0.zip](https://github.com/user-attachments/files/20029816/atmosphere-1.9.0-preprerelease-20_support-9dd8269f7%2Bhbl-2.4.4%2Bhbmenu-3.6.0.zip)
[fusee.bin.zip](https://github.com/user-attachments/files/20029817/fusee.bin.zip)

**Please note:** You should only use this if you are already comfortable with atmosphere, and don't need help with very basic support issues. There may still be issues I don't know about, even though I fixed the only known issue reported with the QA testing build.

BLUE8499, 2025-05-04:
try your pre-release still get unknow pgk1

Rahvin5, 2025-05-05:
First of all thank you for all the hard work you do on this project.

Second, looks like Nintendo broke something with their 20.0.0 release and they've now released 20.0.1 to fix it.

Hopefully this doesn't affect what you've done thus far?

9Bitdo, 2025-05-05:
> First of all thank you for all the hard work you do on this project.
> 
> Second, looks like Nintendo broke something with their 20.0.0 release and they've now released 20.0.1 to fix it.
> 
> Hopefully this doesn't affect what you've done thus far?

https://github.com/Atmosphere-NX/Atmosphere/issues/2502#issuecomment-2849410425

EVIL-BLUE-DRAG0N, 2025-05-05:
THANK FOR ALL YOUR HARD WORK

Pandabug19, 2025-05-05:
Atmosphere crashes after is loaded into switch home screen. Tried to play nsp rom file. System crashes.

SciresM, 2025-05-05:
Piracy is not supported.

Ibim243, 2025-05-05:
When trying to boot into Sysmmc or Stock i get the error:

Unknown pkg1 version 
HOS version not supported!

My system forced me to update to 20.0.0 or 20.0.1
Emummc does still works since it's not updatetd.

nicklo0, 2025-05-05:
> I drag and dropped the new contents of the folder to the root of my sd card and replaced fusee.bin 
> 
> When trying to boot into Sysmmc or Stock I get the error:
> 
> Unknown pkg1 version 
> HOS version not supported!
> 
> My system forced me to update to 20.0.0 or 20.0.1 a few days prior (idk which one, is there a way to chec?)
> Emummc does still works since it's on 19.0.1
> 
> Did I miss a step or what's the issue here?

That happened when i tried to start it with an old hekate version, try the new test version of hecate or inject fusee directly.

Ibim243, 2025-05-05:
> > I drag and dropped the new contents of the folder to the root of my sd card and replaced fusee.bin
> > When trying to boot into Sysmmc or Stock I get the error:
> > Unknown pkg1 version
> > HOS version not supported!
> > My system forced me to update to 20.0.0 or 20.0.1 a few days prior (idk which one, is there a way to chec?)
> > Emummc does still works since it's on 19.0.1
> > Did I miss a step or what's the issue here?
> 
> That happened when i tried to start it with an old hekate version, try the new test version of hekate or inject fusee directly.

I will try

SylonAge, 2025-05-05:
I installed the system on a brand new SD card and it started to crash when connected to the internet. I created a system_settings.ini file for prodinfo blanking as in your message "this isn't an atmosphere issue, you need to disable prodinfo blanking or this error occurs" and wrote 

[blank_prodinfo_sysmmc]
enable_false_prodinfo = u8!0.

 It is now working without crashing, but the nsp I installed does not open saying checking software information. What should I do?

SylonAge, 2025-05-05:
I installed Sigpatches from this link https://github.com/AK478BB/Sigpatches/releases and my problem was solved.

Ibim243, 2025-05-05:
> I will try

It worked! At least for stock. 
Booting into Sysmmc results in the error: 

A fatal error occured when running Athmosphère:
Programm ID: 010000000000bd00
Error Desc: std::abort() called (0xffe)

Report saved to /athmosphere/fatal_errors/reports_0000000008fdb6fc.bin

KTa0ADlHVVHA, 2025-05-05:
works for me, installed patches from https://github.com/impeeza/sys-patch/releases/tag/v1.5.6

Soneoylys, 2025-05-05:
Could you guys stop talking about piracy tools here please, piracy is not supported here and we absolutely hate it, thanks.

SouthPokeyz, 2025-05-05:
dunno wum but I think everyone loves piracy:) what's what would this atmosphere thing be for 😭

olibri-us, 2025-05-05:
> dunno wum but I think everyone loves piracy:) what would this atmosphere thing be for other then to get free games 😭

First of all, piracy can lead to a trial with Nintendo, which they probably don't want.
Second, there are a lots of things to do other than free games.
I don't have any pirate games and I'm still using Atmosphere...

[114 more replies]