Atmosphere-NX/Atmosphere #2503: A fatal error occured when running fuse

cn1t, 2025-04-30:
## Bug Report

### What's the issue you encountered?

I am just now setting up my switch for modding.
When i flashed the fusee.bin file, my switch all of the sudden only displays this:

A fatal error occured when running Fusée
Unable to identify package1!
Press POWER to reboot.

I ran a switch system update before trying to mod it (maybe thats it?)

### Crash Report

The fatal_errors folder is empty

### System Firmware Version

I cant exit this screen and cant even shut the switch down because all it says is reboot and it just reboots to that screen.

### Environment?

Newest release: 1.8.0

masagrator, 2025-04-30:
Atmosphere doesn't support currently 20.0.0 FW. Just wait for new Atmosphere release.

cn1t, 2025-04-30:
> Atmosphere doesn't support currently 20.0.0 FW. Just wait for new Atmosphere release.

Oh... How can i downgrade to an older version then? I could only find tutorials that use atmosphere itself to downgrade the HOS

cn1t, 2025-04-30:
> You can't if you don't want to wipe out whole system and block online functionalities forever. If you have backup you can restore it. Otherwise wait.
> 
> PS. After downgrading you won't be able to use it anyway without CFW

I dont care about online stuff or the data on the switch because its unable to read cards anyway and i barely bought anything on the eshop. How can i wipe the whole system and downgrade?

ELY3M, 2025-04-30:
you have to wait for new atmosphere release.    just never ever downgrade!  it is not worth the risk.   
I am sure it will be a few days.

masagrator, 2025-04-30:
Google "Rebuilding NAND on Nintendo Switch". Remember that's nuclear option and there is no going back.

ELY3M, 2025-04-30:
to exit that screen, hold power button and it will turn off.

cn1t, 2025-04-30:
> you have to wait for new atmosphere release. just never ever downgrade! it is not worth the risk. I am sure it will be a few days.

You think it will be out on saturday? Thats when i need the modded switch

JMenGUI, 2025-04-30:
> You think it will be out on saturday? Thats when i need the modded switch

We are aware that you are a new user but please, show the bare minimum respect for the developer work specially if you are following a guide and not understanding 100% of what you are doing.

Also, what you need to wait for is not Atmosphere but the upcoming version of hekate:
[https://github.com/CTCaer/hekate/issues/1081](url)

closed 2025-04-30 by SciresM