Atmosphere-NX/Atmosphere #2504: Kernal Panic right after booting from Hekate

heydemoura, 2025-04-30:
## Bug Report

### What's the issue you encountered?
- Clean Atmosphere 1.8.0 install;
- Happens right after booting from hekate

```
# hekate_ipl.ini

[config]
autoboot=0
autoboot_list=0
bootwait=3
backlight=100
autohosoff=0
autonogc=1
updater2p=1
bootprotect=0

[CFW emuMMC - Atmosphere]
fss0=atmosphere/package3
emummcforce=1
icon=bootloader/res/icon_payload.bmp

[Stock]
fss0=atmosphere/package3
stock=1
emummc_force_disable=1
icon=bootloader/res/icon_switch.bmp

```

[image]

### How can the issue be reproduced?

Running on Steam Deck OLED, booting from hekate using the above configuration on a clean install

### Crash Report
[report.zip](https://github.com/user-attachments/files/19983274/report.zip)

### System Firmware Version

13.2.0

### Environment?
- Hekate
- Official Release 1.8.0
- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- EmuMMC file-based

heydemoura, 2025-04-30:
Trying Official Release 1.5.4 worked just fine. I will be updating Atomsphere and EmuMMC from this one. Just thought I should contribute at least with the error report.

poudigne, 2025-04-30:
@heydemoura Atmosphere 1.8.0 won't work if you didn't update you system to 18.0.x. Google on how to do it.

closed 2025-04-30 by SciresM