Atmosphere-NX/Atmosphere #2508: I have this problem after upgrading sysnand, with lower emunand

Vorffeed, 2025-05-03:
Hello, this happened to me: I have an OLED Switch with an Instinct V6S chip. I updated the SysNAND to version 20.0.1, and when I try to boot into EmuNAND, after the first image loads, the screen goes black and I get the following message:

"A fatal error occurred when running fusée.
Failed to locate warmboot firmware!
Press power to reboot."

I thought that even if the SysNAND was updated to a high version, it would still be possible to boot into an older EmuNAND without issues, since they are separate systems EmuNAND is on a partition of the microSD isolated from the SysNAND.

blacko3, 2025-05-03:
Are you a .... g-g-g-girl?

Skonikol, 2025-05-03:
https://yadi.sk/d/kCHBc1wIPwNRgg

closed 2025-05-03 by SciresM

rumla3434, 2025-05-04:
https://github.com/rumla3434/Atmosphere-Stable/releases/download/1.9.0/AMS1.9.0-Test-v250505.rar