Atmosphere-NX/Atmosphere #2509: I get an error with the atmosphere screen 2002-3005 0x00177a02

ohheydarkside, 2025-05-04:
## Bug Report

### Crash Report
 [gist](https://gist.github.com/ohheydarkside/81d3ce7a406bce212fad6b471b71d1a1)
### System Firmware Version

20.0.1

### Environment?

- hekate
- EmuMMC partition

borntohonk, 2025-05-04:
> ## Bug Report
> ### Crash Report
> 
> [gist](https://gist.github.com/ohheydarkside/81d3ce7a406bce212fad6b471b71d1a1)
> ### System Firmware Version
> 
> 20.0.1
> ### Environment?
> 
>     * hekate
> 
>     * EmuMMC partition

Disable prodinfo_blanker, undo "incognito", or find a means to patch out the crash condition in NIM.

borntohonk, 2025-05-04:
> > > ## Bug Report
> > > ### Crash Report
> > > [gist](https://gist.github.com/ohheydarkside/81d3ce7a406bce212fad6b471b71d1a1)
> > > ### System Firmware Version
> > > 20.0.1
> > > ### Environment?
> > > ```
> > > * hekate
> > > 
> > > * EmuMMC partition
> > > ```
> > 
> > 
> > Disable prodinfo_blanker, undo "incognito", or find a means to patch out the crash condition in NIM.
> 
> where i can disable it?

You're the one who enabled it, figure it out.