Atmosphere-NX/Atmosphere #2512: connecting to internet causes system crash

falsonix, 2025-05-05:
## Bug Report

**PLEASE don't see this as some person malding over issues in a test build. I only seek to assist in the elimination of bugs and a better overall user experience.**

### What's the issue you encountered?

Downloading the extremely early build of Atmosphere from issue 2502 (_Originally posted by @SciresM in [#2502](https://github.com/Atmosphere-NX/Atmosphere/issues/2502#issuecomment-2849410425)_) has the following issues (for me at least):
- Upon connecting to the Internet, the console crashes
- package3 cannot be used to launch into the system menu, only fusee.bin works

### How can the issue be reproduced?
(this is for the Internet issue, no clue about the other one)

1. Turn on the console using the POWER button
2. Launch Atmosphere via the fusee.bin payload (I'm using Hekate to launch the payload in this case)
3. Allow the console to connect to the Internet

### Crash Report

https://gist.github.com/falsonix/50eb35d822644c5c0dfd9349785cabd6

### System Firmware Version

20.0.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- fusèe
- Official release or unofficial build: 
  - Unofficial build
  - Obtained through a post from Issue #2502 
     - Experimental 20.0.0 build from SciresM
- Do you have additional kips or sysmodules you're loading:
- No
- Homebrew software installed: [ Items included with HATS, so: 
Switch 90DNS tester
Hekate+AMS-package3-sigpatches-1.8.0P2-cfw-19.0.0_V6
hekate_ctcaer_6.2.2_Nyx_1.6.4
Atmosphere-1.8.0-prerelease-c6014b533
Tinfoil 19.0 V1 Applet Mode
Tinfoil 19.0 V1 NSP version (inside Switch/Tinfoil/NSP)
DBI 658
Homebrew Appstore 2.3.2
FTPD Pro 3.1.0
NX-Shell 4.0.1
Tesla Menu (ovlmenu.ovl) 16.00
JKSV 11/05/2024
Edizon SE 3.8.37
AIO-Switch-Updater V2.23.2
Goldleaf 1.0.0
Mission Control 0.12.0
Linkalho 2.0.1
Tinwoo 1.0.26 ]
(sorry if that is a list too long, just putting what the site says)
- EmuMMC or SysNAND:
  - SysNAND

### Additional context?

- Additional info about your environment:
- Just installed a HWFly chip into my console earlier today and was excited to test Homebrew, but I had already updated to 20.0.0 so that was not possible at the time

apologies if i did something wrong, i'm still very new to all of this >-<

falsonix, 2025-05-05:
I also recorded two instances of this happening on my system:

https://www.youtube.com/watch?v=iO1nVIZ4Zrg

jvrcruzGAMES, 2025-05-05:
I managed to get to HB menu with the pre-pre-release, When I added sys-patch, I got the same problem...
It is definitly not stable and there are some things that need to be fixed.

SciresM, 2025-05-05:
@falsonix this isn't an atmosphere issue, you need to disable prodinfo blanking or this error occurs. And piracy not supported.

@jvrcruzGAMES the problem is literally your piracy patches. Nothing needs to be changed about ams there.