Atmosphere-NX/Atmosphere #2513: Building New code error

impeeza, 2025-05-05:
## Bug Report

Sorry for bother, I hope you can help on this issue.

### What's the issue you encountered?

Building the code for the branch "20_support" ends with the error:

[image]

I am using Ubuntu and MSYS's MinGW64 on both OSs is the same error.

I am also using the LibNX's branch 20_support.

I know there is so much keeping you busy but this could be helpful in the nearest future.

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]
Install devkitpro and all prerequisites.
build and instal LibNX 20_support branch
clone Atmosphere repo
run make

### Crash Report

I think somewhere on the make files a parameter for GCC indicating the need to use GNU++23 is missing.

SciresM, 2025-05-05:
Atmosphere cannot be built externally, right now. Please wait for a public release of the relevant compilers.

closed 2025-05-05 by SciresM

impeeza, 2025-05-05:
Thanks mate,  and thanks for all your great work.