Atmosphere-NX/Atmosphere #2515: Error code 2168-002 (0x4a8) on pre-release

julianblack, 2025-05-08:
## Bug Report

### What's the issue you encountered?

Firmware: 20.0.1

When I lauch sys cfw I get an error screen displaying:
Error code 2168-0002 (0x4a8)

Right after the switch logo.  

### How can the issue be reproduced?

I've updated to the pre-release seen below:
atmosphere pre prerelease: https://github.com/user-attachments/files/20029816/atmosphere-1.9.0-preprerelease-20_support-9dd8269f7%2Bhbl-2.4.4%2Bhbmenu-3.6.0.zip
hekate prerelease: https://github.com/user-attachments/files/20029984/hekate_20.0.0_support_test.zip

Copied in the extract to the root of my SD card + bootloader from hecktate

### Crash Report

[01746742487_0100000000001000.log](https://github.com/user-attachments/files/20111503/01746742487_0100000000001000.log)

### System Firmware Version

20.0.1

### Environment?

heckate attached link above

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: just smash mods
- EmuMMC or SysNAND:
  - sysNAND

### Additional context?

- Additional info about your environment:

Aura67, 2025-05-08:
0100000000001000 is a custom theme delete 0100000000001000 from your contents folder under atmosphere and then try to boot again the tools also have to be adapted for version 20.0.0 again just because there is a pre pre-realses doesn't mean that the other developers have already updated their tools for it and there isn't even a pre-realses out the last stable is for version 19.0.1

smashplayer91, 2025-05-08:
> 0100000000001000 is a custom theme delete 0100000000001000 from your contents folder under atmosphere and then try to boot again the tools also have to be adapted for version 20.0.0 again just because there is a pre pre-realses doesn't mean that the other developers have already updated their tools for it and there isn't even a pre-realses out the last stable is for version 19.0.1 and as it seems to me your message is not an atmosphere problem.

Ah you are the best.  Fixed the issue!

closed 2025-05-08 by smashplayer91