Atmosphere-NX/Atmosphere #2516: Error while attempting to build atmosphere( 20.0.0)

A1nitaK-SG, 2025-05-09:
encountered an error while building atmosphere (20.0.0) on MSYS2. seems like some file are missing ?
all packages installed including lz4

In file included from C:/Users/DT/repos/Atmosphere-20_support/Atmosphere-20_support/fusee/program/source/mtc/fusee_mtc_erista.cpp:35:
C:/Users/DT/repos/Atmosphere-20_support/Atmosphere-20_support/fusee/program/source/mtc/fusee_mtc_tables_erista.inc:18:12: fatal error: ../../mtc_tables/combined/T210SdevEmcDvfsTableS6gb01/table.bin: No such file or directory
   18 |     #embed "../../mtc_tables/combined/T210SdevEmcDvfsTableS6gb01/table.bin"
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[4]: *** [/opt/devkitpro/devkitARM/base_rules:34: fusee_mtc_erista.o] Error 1
make[3]: *** [/c/Users/DT/repos/Atmosphere-20_support/Atmosphere-20_support/fusee/program/program.mk:54: all] Error 2
make[2]: *** [/c/Users/DT/repos/Atmosphere-20_support/Atmosphere-20_support/fusee/fusee.mk:29: check_program] Error 2
make[1]: *** [/c/Users/DT/repos/Atmosphere-20_support/Atmosphere-20_support/atmosphere.mk:127: fusee] Error 2
make[1]: Leaving directory '/c/Users/DT/repos/Atmosphere-20_support/Atmosphere-20_support'
make: *** [Makefile:39: nx_release] Error 2

borntohonk, 2025-05-09:
> T210SdevEmcDvfsTableS6gb01

Seems update_mtc_tables.py isn't being ran, due to it being a python2 scripts, which is deprecated.

and that'd be what produces whatever file you're missing.

```py
write_file(os.path.join('mtc_tables/combined/%s' % board, 'table.bin'), data)
```

SciresM, 2025-05-09:
I'll just add the binary files to the repo later, since they're now #embed output. 

@borntohonk the inc files are supposed to be script output and always have been.

closed 2025-05-09 by SciresM