Atmosphere-NX/Atmosphere #2519: 2168-0002 crash after updating to 20.0.1

Conrae, 2025-05-09:
Recently updated my sysmmc to 20.0.1 via Daybreak and made sure to update atmosphere and hekate. I know the most common reason for this error is custom themes, which I uninstalled prior to updating. I also deleted the folder in atmosphere/contents that contains themes. I tried to delete 010000000000000f since the error states that's the folder the issue is coming from, but I do not have that folder in atmosphere/contents and am unable to find it elsewhere. The crash report has been attached.

[01746833486_010000000000000f.log](https://github.com/user-attachments/files/20130294/01746833486_010000000000000f.log)

TLNiles, 2025-05-10:
Having a similar issue here. After updating the system firmware and Atmosphere, the system crashes right on load.
As of note, the crash happens when console connects to wifi/internet. If a console stays in Flight Mode, everything seems to work normally.

[01746844488_010000000000000f.log](https://github.com/user-attachments/files/20132812/01746844488_010000000000000f.log)

SavXL, 2025-05-10:
> Having a similar issue here. After updating the system firmware and Atmosphere, the system crashes right on load. As of note, the crash happens when console connects to wifi/internet. If a console [...]

I second this. Having the same issue where a crash will happen upon connecting to the internet. Putting the switch in airplane mode keeps this from happening. I wonder why this is. I opened the log, but I have too little experience to know what any of it means. Hopefully it'll be useful!

[01746845704_010000000000000f.log](https://github.com/user-attachments/files/20132892/01746845704_010000000000000f.log)

rayxiin, 2025-05-10:
Atmosphere/contents
Delete the folder withe long number.

RSSPLOTER, 2025-05-10:
> Having a similar issue here. After updating the system firmware and Atmosphere, the system crashes right on load. As of note, the crash happens when console connects to wifi/internet. If a console [...]

You can try this https://github.com/Atmosphere-NX/Atmosphere/issues/2463#issuecomment-2509150084 (at your own risk)

borntohonk, 2025-05-10:
> 01746833486_010000000000000f.log

If anyone else having this exact nifm crash (read your logs)

then this applies to you:
Delete "sys-patch",  I.E: "/Atmosphere/contents/420000000000000B" folder  

> You can try this [#2463 (comment)](https://github.com/Atmosphere-NX/Atmosphere/issues/2463#issuecomment-2509150084) (at your own risk)

This is completely unrelated to the pasta penne situation

borntohonk, 2025-05-10:
This should be closed, as it's not an atmosphere issue.

Conrae, 2025-05-10:
> > 01746833486_010000000000000f.log
> 
> If anyone else having this exact nifm crash (read your logs)
> 
> then this applies to you: Delete "sys-patch", I.E: "/Atmosphere/contents/420000000000000B" folder
> 
> (it is what is causing this specific crash)
> 
> > You can try this [#2463 (comment)](https://github.com/Atmosphere-NX/Atmosphere/issues/2463#issuecomment-2509150084) (at your own risk)
> 
> This is completely unrelated to the pasta penne situation

This fixed it, you're a lifesaver. I have no idea how to read these crash logs so I wasn't certain if the issue originated from Atmosphere or not. I'll be closing this now.

closed 2025-05-10 by Conrae

HandsomeKai462, 2025-05-12:
So I deleted the folder, and I'm still having the same problem when I connect to the internet :(