Atmosphere-NX/Atmosphere #2520: Error starting emunand on 19.0.0 with sysnand on 20.0.1

Vorffeed, 2025-05-10:
Sorry but I still have the same problem as before, I have updated atmosphere and hekate to the new versions, but when I try to enter emunand which I have on 19.0.0 with sysnand updated to 20.0.1 I keep getting this message, what am I doing wrong ><

[image]

SciresM, 2025-05-10:
You need to boot the firmware version matching your fuses with atmosphere at least once in order to do wake/sleep. Boot 20.0.1 (presumably) at least once.

closed 2025-05-10 by SciresM

Vorffeed, 2025-05-10:
I ask, because I have done just that, I have put in sysnand 20.0.1 and even deleted the warmboot and put in sysnand and it still tells me the same thing.

although I do not know if you mean to enter sysnand, ie in cfw mode because I enter in real mode by disabling the chip and again I tried giving reboot with hekate and selecting ofw and still doing the same ><

SciresM, 2025-05-10:
"boot the firmware version matching your fuses with atmosphere", I obviously mean sysmmc with cfw