Atmosphere-NX/Atmosphere #2521: lto-wrapper.exe error while attempting to make atmosphere

9Bitdo, 2025-05-10:
Encountered an issue while building latest atmosphere. I would like to seek some advice what am i missing. 

Unsure if Lz4 library is root of the issue but attempt to reinstall Lz4 gave me
" Requirement already satisfied: Lz4 in c:\python27\lib\site-packages (2.2.1)
Requirement already satisfied: future in c:\python27\lib\site-packages (from Lz4) (1.0.0) "

lto-wrapper.exe: warning: using serial compilation of 2 LTRANS jobs
lto-wrapper.exe: note: see the '-flto' option documentation for more information
C:/devkitPro/devkitARM/bin/../lib/gcc/arm-none-eabi/15.1.0/../../../../arm-none-eabi/bin/ld.exe: warning: section `.main.fill' type changed to PROGBITS
C:/devkitPro/devkitARM/bin/../lib/gcc/arm-none-eabi/15.1.0/../../../../arm-none-eabi/bin/ld.exe: warning: C:/Users/9Bitdo/Atmosphere/fusee/program/out/nintendo_nx_arm_armv4t/release/program.elf has a LOAD segment with RWX permissions
arm-none-eabi-objcopy -S -O binary --set-section-flags .bss=alloc,load,contents /c/Users/9Bitdo/Atmosphere/fusee/program/out/nintendo_nx_arm_armv4t/release/program.elf /c/Users/9Bitdo/Atmosphere/fusee/program/out/nintendo_nx_arm_armv4t/release/program.bin
built ... program.bin
bash: line 1: python: command not found
make[4]: *** [/c/Users/9Bitdo/Atmosphere/fusee/program/program.mk:87: /c/Users/9Bitdo/Atmosphere/fusee/program/out/nintendo_nx_arm_armv4t/release/program.lz4] Error 127
make[3]: *** [/c/Users/9Bitdo/Atmosphere/fusee/program/program.mk:54: all] Error 2
make[2]: *** [/c/Users/9Bitdo/Atmosphere/fusee/fusee.mk:29: check_program] Error 2
make[1]: *** [/c/Users/9Bitdo/Atmosphere/atmosphere.mk:127: fusee] Error 2
make[1]: Leaving directory '/c/Users/9Bitdo/Atmosphere'
make: *** [Makefile:39: nx_release] Error 2

Vorffeed, 2025-05-10:
It is telling you that it can't find python, have you added python to the path?

9Bitdo, 2025-05-10:
C:\Windows\System32>python --version
Python 2.7.18

Vorffeed, 2025-05-10:
I mean the whindows path, if you have added there, the path to the python folder.
I refer to the whindows path, that if you have there added, the path to the python folder.
In start, look for edit windows environment variable, in the window that opens, advanced options tab, last option, environment variables, in the new window that opens, look for path, click on edit, and there you add the path of the python folder.

9Bitdo, 2025-05-10:
yea you wont be able to use python in elevated command prompt if you had not added the python path to the user enviroment variables. i did the same for pip by adding python/scripts to it as well

Vorffeed, 2025-05-10:
I asked you because it says bash: line 1: python: command not found, that's why I asked you if you had added to the windows python path.

9Bitdo, 2025-05-10:
> I asked you because it says bash: line 1: python: command not found, that's why I asked you if you had added to the windows python path.

i know and thats why i answered your question

Vorffeed, 2025-05-10:
The only thing I can think of is to try to install the latest version of phython.

9Bitdo, 2025-05-10:
> The only thing I can think of is to try to install the latest version of phython.

are you on the latest python ?

Vorffeed, 2025-05-10:
I'm in the previous Python 3.13.2

closed 2025-05-10 by SciresM