Atmosphere-NX/Atmosphere #2522: Error Code Related Stuff - 0xCA7C (2124-0101)

maybekoi, 2025-05-11:
Got this while developing a homebrew.

### Crash Report

https://github.com/maybekoi/U-CrashLogs/blob/main/Atmosphere/log.log

### System Firmware Version

19.0.1

### Environment?

- hekate
- SysNAND

SciresM, 2025-05-11:
> account::ResultUserAlreadyOpened()

Indicates this is a problem with the homebrew, not an atmosphere problem.

On closer inspection this whole report is junk, program id is invalid (indicates piracy/installed content? not supported, maybe ams bug causes wrong program id in report), also doesn't even appear to be homebrew, crash report mentions nnSdk + unity SwitchPlayer.nss.

Anyway in any case this isn't an atmosphere issue.

closed 2025-05-11 by SciresM

maybekoi, 2025-05-11:
hm..alright thanks anyways