Atmosphere-NX/Atmosphere #2524: error 0x4A8 (2168-0002)  after update to 20.0.1

Valou5231, 2025-05-11:
**Atmosphère Crash Report (v1.7):
Result:                          0x4A8 (2168-0002)

Process Info:
    Process Name:                nifm
    Program ID:                  010000000000000f
    Process ID:                  000000000000006a
    Process Flags:               00001137
    User Exception Address:      0000003ff660a000
Exception Info:
    Type:                        Data Abort
    Address:                     39400c0854000291
    Fault Address:               0000000000000000
Crashed Thread Info:
    Thread ID:                   00000000000001c9
    Stack Region:                0000002681f5d000-0000002681f6d00**

after i update my switch to firmware 20.0.1 with atmosphere pre-realese my switch crash when i try to **lauch** atmsphere . plz help me

AleosUnderscore, 2025-05-11:
Same, i'm on 20.0.1 too and it crashes after some seconds in main menu.

my crash log if it helps

[01746979485_010000000000000f.log](https://github.com/user-attachments/files/20150148/01746979485_010000000000000f.log)

SciresM, 2025-05-11:
Piracy is not supported

closed 2025-05-11 by SciresM