Atmosphere-NX/Atmosphere #2528: After connecting to Wi-Fi, the power consumption becomes abnormal.

do-kiss, 2025-05-12:
Once Wi-Fi is turned on, abnormal power consumption occurs — even in sleep mode, it consumes about 8% per hour. Currently, the Switch firmware version is 20.0.1, and the Atmosphère version is 1.9.0. No other plugins are used; it is a completely clean Atmosphère environment. If I enable Airplane Mode, the abnormal power consumption issue disappears.

SciresM, 2025-05-12:
There have been some reports of this, yes. I don't really know what it could be, besides some interaction with dns.mitm, but people have said that even disabling dns.mitm doesn't reduce drain. It would be helpful for you to confirm that.

91ajames, 2025-05-13:
I did a test my self, putting on stand by on Emu, no game open, even though it doesn't work to wake it, in the last 14 1/2 hrs (est 12am to 7:20PM) from 100% to 28.9%. and this is with the wifi left on.

In the last 10mins on Display Hekate v6.2.2 boot screen, -226mA to 274mA, at a loss 1.0% drain, 27.9%. Surely this is not normal, I don't remember it going past 234Am from the last version. 

- Oled Zelda Switch Edition - 18.0.1|AMS 1.9.0|E

Incase this helps. I accidentally upgraded my software so its up to date. 20.0.1

SciresM, 2025-05-13:
@91ajames Every indication so far is that this affects OFW as well as CFW, and isn't actually related to Atmosphere per-se.

do-kiss, 2025-05-14:
> [@91ajames](https://github.com/91ajames) Every indication so far is that this affects OFW as well as CFW, and isn't actually related to Atmosphere per-se.

Actually, I also tested the official stock system without loading Atmosphère, and the official 20.0.1 version of the system showed no abnormal power consumption — even when connected to Wi-Fi, there was no excessive drain. The power drain issue only occurs when running Atmosphère in emuMMC.

SciresM, 2025-05-14:
@do-kiss this is inconsistent across setups and has to do with various wireless settings. Some people see it with atmosphere and don't without, because of various settings differences/wireless usage. Some people see it on completely unhacked consoles. Some people see it depending on 2.4 vs 5ghz network. It's complicated.

do-kiss, 2025-05-14:
> [@do-kiss](https://github.com/do-kiss)这在不同设置之间是不一致的，并且与各种无线设置有关系。由于各种设置差异/无线使用，有些人用气氛看，而不用气氛看。有些人在完全未被黑客入侵的控制台上看到了它。有些人看它取决于2.4与5ghz网络。这很复杂。

> [@do-kiss](https://github.com/do-kiss) this is inconsistent across setups and has to do with various wireless settings. Some people see it with atmosphere and don't without, because of various [...]
Yes, but I’ve tested everything I could — including Bluetooth, NFC, 2.4GHz/5GHz Wi-Fi, etc. I can confirm that on my emuMMC, simply being connected to Wi-Fi causes the battery drain. However, if I delete the internet settings in the system settings, the issue goes away even without enabling airplane mode. This suggests the power drain is solely related to being connected to Wi-Fi. I’m using the Hekate + Atmosphère + Sys-patch trio setup.I’ve also asked others around me to run the same tests, and they encountered the same battery drain issue. This makes me wonder if the cause might be related to Hekate or Sys-patch.

SciresM, 2025-05-14:
Piracy is not supported, have to block you from the repo, sorry. 

Anyway, yes, it's known that the issue is "Nintendo wireless code wakes up the console too often, causing battery drain" since 18.0.0, but it's not well understood what exactly triggers this. Again, it's likely unrelated to atmosphere/hekate/etc. It occurs without atmosphere for many people. It is likely just a byproduct of your different emummc environment.

AK478BB, 2025-05-16:
> I did a test my self, putting on stand by on Emu, no game open, even though it doesn't work to wake it, in the last 14 1/2 hrs (est 12am to 7:20PM) from 100% to 28.9%. and this is with the wifi left [...]

you can make a test with two emunands in one console  and compare the power consumption.  
1. the first emunand SD00,  update from 19.0.1 or lower to 20.0.1 by daybreak.   still old news in channel.
2. the second emunand SD01, new create with offical firmware by 20.0.1.  the latest news in channel.

muxi1, 2025-05-16:
I have so far rebuilt two consoles (Erista on sysNand 20.0.1 and OLED on sysNand 18.0.1) with a partition emuMMC. In the end, both consoles were on emuMMC 20.0.1, whereby I first had to update the OLED to the latest version via Daybreak.
Both installations solved the problem with the rapid battery drain in standby! However, another test with an Erista on sysNand 1.0.0 did not work. I am currently testing this under a Switch Lite on sysNand 18.0.0.

Its-my-right, 2025-05-16:
On Sys (20.0.1) Stock with internet access (wifi) and no DNS blocking, I dont have the issue.
On Sys (20.0.1) CFW with internet access (wifi) and no DNS blocking, I dont have the issue.
On Emu (20.0.1, upgraded via daybreak from 19.0.0) with internet access (wifi) and DNS blocking, I have the issue.
On Emu (20.0.1, made from sys 20.0.1 after this sys had access to internet without DNS blocking) with internet access (wifi) and DNS blocking, I don't have the issue.

It is pure speculation because I don't know at all how to check this, but I am starting to think, maybe after a firmware update, if internet is connected, the console tries to "acquit" something from N servers. Obviously it can't if you upgraded an EmuNand and have internet access BUT DNS blocking. So maybe it would explain the battery drain because of infinites tries ?

Konzolista, 2025-05-17:
So now I don't know if this is a new thing or if you already know it and someone tried it. I did it on V1 RCM.
I did this test. I had emuNAND and sysNAND version fw 13.0.1. First I updated OFW sysNAND to 20.0.1. Then I removed everything "unclean" from the card and uploaded only clean AMS 1.9.0. and Hekate 6.3.0. So no patches, no sig or sys, no exosphere.ini, no default.txt in hosts, as I write, only clean AMS, just to be sure. When emuNAND started up to fw 13.0.1 I connected to wifi and updated the fw officially from N servers to 20.0.1. After the reboot, Hekate autostarted to emuNAND, then the AMS logo, then the N logo and then the swich did not boot into the system but rebooted itself again. I have the impression that it does this now after all the OFW updates to 20.0.1 (I had not tried 20.0.0 before). Then everything booted normally to emumNAND 20.0.1 AMS 1.9.0E. I turned off the console and added everything that is used and recommended for a safe emuNAND to the mSD, so sys_p, default.txt to host and exosphere.ini. Everything according to the recommendations of "Setup Exosphere / DNS MITM". Do you know what is happening now? No blocking of N servers!!! I get to Shop N, to Online N, as if the system did not block any DNS N! It scares me!!!
And now someone explain to me why it is behaving like this now. The official Update system is obviously doing something different or in addition to what Daybreak does. I haven't checked the consumption yet, it scares me that the console can now access the N server without any problems. And by the way, my sn number is XAW0000000...
Google translate...

luuk00000000000000000000000000000000000, 2025-05-21:
People in gbatemp's Atmosphere thread are saying that deleting and re-creating the old (I assume pre-20.x.x update) profiles seems to fix the battery drain issue.

I haven't tested this myself but it is at least a bit less extreme than just nuking and rebuilding your emuMMC.

johnkelvinliu, 2025-05-28:
I think only some online traffic would not caused such high power consumption heat problem. The heat is just like after after playing games for a period of time.

johnkelvinliu, 2025-05-28:
> So now I don't know if this is a new thing or if you already know it and someone tried it. I did it on V1 RCM. I did this test. I had emuNAND and sysNAND version fw 13.0.1. First I updated OFW sysNAND to 20.0.1. Then I removed everything "unclean" from the card and uploaded only clean AMS 1.9.0. and Hekate 6.3.0. So no patches, no sig or sys, no exosphere.ini, no default.txt in hosts, as I write, only clean AMS, just to be sure. When emuNAND started up to fw 13.0.1 I connected to wifi and updated the fw officially from N servers to 20.0.1. After the reboot, Hekate autostarted to emuNAND, then the AMS logo, then the N logo and then the swich did not boot into the system but rebooted itself again. I have the impression that it does this now after all the OFW updates to 20.0.1 (I had not tried 20.0.0 before). Then everything booted normally to emumNAND 20.0.1 AMS 1.9.0E. I turned off the console and added everything that is used and recommended for a safe emuNAND to the mSD, so sys_p, default.txt to host and exosphere.ini. Everything according to the recommendations of "Setup Exosphere / DNS MITM". Both the new Goldleaf 1.1.0 and TegraExplorer 4.2.0 made the same fw dump from emuNAND but be careful, there are no longer 235 files like you do from OFW sysNAND but only 231 files! Maybe after that mysterious console reboot after updating emuNAND, four files were deleted. I don't understand the reason, I'm not an expert. The dump with 231 files is not valid according to Goldleaf 1.1.0 and will not allow it to be installed. But the console works. I updated the fw emuNAND via Daybreak but with a valid dump of 235 files and the console also works after the update. I tried to update emuNAND again with the invalid dump with 231 files via Daybreak. This warning popped up but it allowed me to update. The console works. Update is valid! exFAT Validation failed with result: 0x00000202 Missing content: -Program id: 010000000000081c -Version: 20.0.0 -Content id 00000000000000000000000000000000
> 
> mSD is formatted as FAT32, so maybe it's just related to that, I don't know... I'll go test the consumption with the update with 231 files and write it here.... Google translate...
> 
> EDIT: My conclusion: The battery consumption with WiFi on is not affected by the update method. Updating via Daybreak or via wifi via N servers makes no difference on the V1 RCM console in terms of battery consumption. The only difference is that via N servers not all 235 files are saved to the system, but only 231. This does not affect the functionality of the console if FAT32 is used on the mSD. During the update via N servers, exFAT support files are damaged or not saved. It is questionable how the update would turn out if the mSD card was formatted to exFAT, maybe all 235 update files would be saved to the system. But I can't find out, I don't use exFAT...

I also have this problem during installing a firmware from sysNAND after updated officially:
exFAT Validation failed with result: 0x00000202
Missing content:
-Program id: 010000000000081c
-Version: 20.0.0
-Content id 00000000000000000000000000000000

Everytime need to download firmware from the web site instead to prevent this from happening.
However it only happen in one of my switch OLED while the other switch OLED and switch lite do not have such problem.
It happens in every dumps of previous and current firmwares.
I doubt whether it is caused by any problem occured/damaged sysNAND in the MOD chip installation.

Konzolista, 2025-06-01:
> > Tak teraz neviem, či je to nová vec, alebo ak už viete to niekto skúsil. Ja som to urobil na V1 RCM. Urobil som tento test. Mal som emuNAND a sysNAND verzia fw 13.0.1. Najprv som aktualizované OFW sysNAND na 20.0.1. Potom som odstránil všetko, čo "nečistý" z karty a odovzdať iba čisté AMS 1.9.0. a Hekate 6.3.0. Takže žiadne fľaky, žiadne sig alebo sys, č exosphere.ini, č. default.txt v hostiteľov, ako som napísal, len čisté AMS, len pre istotu. Keď emuNAND začal až fw 13.0.1 som pripojený na wifi a aktualizovať fw oficiálne z N servery na 20.0.1. Po reštarte, Hekate autostarted na emuNAND, potom AMS logo, potom N logo a potom swich nemali zaviesť systém, ale systém reštartuje sám znova. Mám dojem, že je to teraz, po všetkých OFW aktualizácie 20.0.1 (nemal som sa snažil 20.0.0 pred). Potom všetko iba normálne emumNAND 20.0.1 AMS 1.9.0 E. som sa obrátil mimo konzoly a pridal všetko, ktorý je používaný a odporúčaný pre bezpečné emuNAND na mSD, tak sys_p, default.txt na hostiteľa a exosphere.ini. Všetko podľa odporúčaní "Setup Exosphere / DNS MITM". Nové Goldleaf 1.1.0 a TegraExplorer 4.2.0 vyrobené rovnakým fw výpis z emuNAND, ale buďte opatrní, tam sú už 235 súbory, ako si urobiť z OFW sysNAND ale iba 231 súbory! Možno po tomto tajomnom konzoly reštartu po aktualizácii emuNAND, štyri súbory boli odstránené. Nechápem dôvod, nie som odborník. Skládka s 231 súbory nie je platný podľa Goldleaf 1.1.0 a nedovolí, aby ju nainštalovať. Ale konzola funguje. Aktualizoval som fw emuNAND cez Svitaním, ale s platným výpisom z 235 súbory a konzoly funguje aj po aktualizácii. Snažil som sa aktualizovať emuNAND opäť s neplatné dump s 231 súborov cez Úsvite. Toto upozornenie vyskočila, ale to mi umožnilo aktualizáciu. Konzola funguje. Aktualizácia je platná! exFAT zlyhania Overenia s výsledkom: 0x00000202 Chýbajúci obsah: -Program id: 010000000000081c -Verzia: 20.0.0 -id Obsahu 00000000000000000000000000000000 
> > mSD je naformátované ako systém súborov FAT32, takže možno je to len vzťahujúce sa na to, že ja neviem... ja pôjdem test spotreby s aktualizáciou, s 231 súbory a napíšte ho sem.... Google translate... 
> > EDIT: Môj záver: spotrebu batérie s WiFi nie je ovplyvnená metódu aktualizácie. Aktualizácia cez Svitania alebo cez wifi cez N serverov je žiaden rozdiel, o V1 RCM konzoly, pokiaľ ide o spotrebu batérie. Jediný rozdiel je, že cez N servery nie všetky 235 súbory sa uložia do systému, ale iba 231. Nemá to vplyv na funkčnosť konzoly ak FAT32 sa používa na mSD. Počas aktualizácie cez N servery, exFAT podporné súbory sú poškodené alebo nie sú uložené. Je otázne, ako aktualizovať dopadne, ak mSD karta bola naformátovaná na exFAT, možno, že všetky 235 aktualizovať súbory by mali byť uložené do systému. Ale nemôžem zistiť, nechcem použiť exFAT...
> 
> Mám tiež tento problém počas inštalácia firmware od sysNAND po aktualizované oficiálne v jednom z mojich Prepínač OLED: exFAT zlyhania Overenia s výsledkom: 0x00000202 Chýbajúci obsah: -Program id: 010000000000081c -Verzia: 20.0.0 -Obsah id 00000000000000000000000000000000
> 
> Zakaždým, keď je potrebné stiahnuť firmware z webovej stránky namiesto toho, aby sa to nestalo. Však to môže stať iba v jednom z mojich prepínač OLED zatiaľ čo iný prepínač OLED a prepínač lite nemajú taký problém. To sa deje v každej skládok predchádzajúci a aktuálny fw. Som na pochybách, či je to spôsobené akýmkoľvek problémom, vznikali/poškodené sysNAND počas mod chip inštalácie.

I haven't tried it, but I have a hypothesis. If you're running emuNAND from a card, it probably can't update the drivers on the mSD. And since it recommends using FAT32 and not exFAT, that could also be the reason why files with exFAT support can't get into the system....

Konzolista, 2025-06-01:
> > So now I don't know if this is a new thing or if you already know it and someone tried it. I did it on V1 RCM. I did this test. I had emuNAND and sysNAND version fw 13.0.1. First I updated OFW sysNAND to 20.0.1. Then I removed everything "unclean" from the card and uploaded only clean AMS 1.9.0. and Hekate 6.3.0. So no patches, no sig or sys, no exosphere.ini, no default.txt in hosts, as I write, only clean AMS, just to be sure. When emuNAND started up to fw 13.0.1 I connected to wifi and updated the fw officially from N servers to 20.0.1. After the reboot, Hekate autostarted to emuNAND, then the AMS logo, then the N logo and then the swich did not boot into the system but rebooted itself again. I have the impression that it does this now after all the OFW updates to 20.0.1 (I had not tried 20.0.0 before). Then everything booted normally to emumNAND 20.0.1 AMS 1.9.0E. I turned off the console and added everything that is used and recommended for a safe emuNAND to the mSD, so sys_p, default.txt to host and exosphere.ini. Everything according to the recommendations of "Setup Exosphere / DNS MITM". Both the new Goldleaf 1.1.0 and TegraExplorer 4.2.0 made the same fw dump from emuNAND but be careful, there are no longer 235 files like you do from OFW sysNAND but only 231 files! Maybe after that mysterious console reboot after updating emuNAND, four files were deleted. I don't understand the reason, I'm not an expert. The dump with 231 files is not valid according to Goldleaf 1.1.0 and will not allow it to be installed. But the console works. I updated the fw emuNAND via Daybreak but with a valid dump of 235 files and the console also works after the update. I tried to update emuNAND again with the invalid dump with 231 files via Daybreak. This warning popped up but it allowed me to update. The console works. Update is valid! exFAT Validation failed with result: 0x00000202 Missing content: -Program id: 010000000000081c -Version: 20.0.0 -Content id 00000000000000000000000000000000
> > mSD is formatted as FAT32, so maybe it's just related to that, I don't know... I'll go test the consumption with the update with 231 files and write it here.... Google translate...
> > EDIT: My conclusion: The battery consumption with WiFi on is not affected by the update method. Updating via Daybreak or via wifi via N servers makes no difference on the V1 RCM console in terms of battery consumption. The only difference is that via N servers not all 235 files are saved to the system, but only 231. This does not affect the functionality of the console if FAT32 is used on the mSD. During the update via N servers, exFAT support files are damaged or not saved. It is questionable how the update would turn out if the mSD card was formatted to exFAT, maybe all 235 update files would be saved to the system. But I can't find out, I don't use exFAT...
> 
> I also have this problem during installing a firmware from sysNAND after updated officially in one of my Switch OLED: exFAT Validation failed with result: 0x00000202 Missing content: -Program id: 010000000000081c -Version: 20.0.0 -Content id 00000000000000000000000000000000
> 
> Everytime need to download firmware from a web site instead to prevent this from happening. However it only happen in one of my switch OLED while the other switch OLED and switch lite do not have such problem. It happens in every dumps of previous and current firmwares. I doubt whether it is caused by any problem occured/damaged sysNAND during the mod chip installation.

I haven't tried it, but I have a hypothesis. If you're running emuNAND from a card, it probably can't update the drivers on the mSD. And since it recommends using FAT32 and not exFAT, that could also be the reason why files with exFAT support can't get into the system.

Reistoge, 2025-06-01:
> Observation: While being affected by the power drain bug, I can still use my batteryless joy-cons attached to the console, so I assume that there is a process hanging in the background which doesn't allow it to switch to the standby state and just turns off the screen, like a wake-lock. Before the update there were only very few instances of their Home-Button being able to wake the console (both OFW and CFW). I updated from 1.8.0 / 19.0.0 to 1.9.1 / 20.1.0. Edit: Can confirm that creating a new user, transferring all saves and deleting the old one solves the issue.

I tried the recommended workaround of deleting the old user and creating a new one (used DBI to delete, then System Settings to create a new profile), but unfortunately the battery drain issue still persists on my modded OLED Switch. I'm on firmware 20.1.0 with Atmosphère 1.9.1.

Even after transferring all save data to the new user, the device continues to stay warm in sleep mode and the battery drains unusually fast — similar to the behavior described with the wake-lock-like condition.

Has anyone had different results on OLED specifically, or is there an additional step I might be missing?

johnkelvinliu, 2025-06-01:
It looks like upgrading to v20.1.0 firmware+compatible atm & hekate & sigpatch, the hot charging problem is solved.

Switching off the 2 Switch oled +1 switch lite before charging, looks like no hot charging anymore.

Testing sleep charging to see if there is any more problem.

Fredhoob, 2025-06-14:
FWIW, I just upgrade one of my V1 Switches to 20.0.1 and left it in sleep mode for one hour and a half and there was no battery drain, same percentage when I turned it back on. Still not sure if that matters or not but I only have fake NNID's linked with Linkalho.

johnkelvinliu, 2025-06-15:
I and the other one have tried deleting and recreating a player account in emuNAND OS however it did not help. They are stated above.

Fredhoob, 2025-06-15:
> I and the other one have tried deleting and recreating a player account in emuNAND OS however it did not help. They are stated above too.

What Switch models do you have?

johnkelvinliu, 2025-06-15:
3 oled 1 lite

johnkelvinliu, 2025-06-15:
> OK, I just updated my second Switch (same config, Emunand) to 20.0.1 and this one has the battery drain issue, about 5% per hour. The only meaningful difference between the two is that I had to create a new user at a point on the one that does not have the battery drain because the cloud save was copied from the stock. So theoretically, it should resolved by creating another user? I'll try it out.

How did you copy the stock save to emuNAND?
Rebuild emmuNAND?

Fredhoob, 2025-06-15:
> How did you copy the stock save to emuNAND? Rebuild emmuNAND?

Oh no it is an Emunand copies a while ago when stock was still on firmware 16, I think. That's another difference with the V1 Switch which does not have the battery drain, that one was copied from firmware 19.0.1 on stock, so maybe there is some kind of residue carried over from earlier firmware versions on Emunand when you update it from earlier Firmwares, no idea.

I will just rebuild my Emunand from the current stock firmware 20.1.1, it's a bit annoying to have to reinstall the games and updates and restore all the saves but if it consistently solves the issue, great!

samfadel, 2025-06-20:
Recently I update my Nintendo Switch V2 on Firmware 20.1.1 From 19.0.1.

I noticed that my battery after that was drain very fast in sleep mode (wifi on) but when put it in airplane mode there is no battery drain.
I have install Atmosphere 1.9.1 and up Hekate 6.3.1.
Everything is fine with the games.

masagrator, 2025-06-20:
> It's possible that certain background processes in Atmosphere

And did you consider that maybe it's not Atmosphere process but some other process that you have put there and it's running in background?

Fredhoob, 2025-06-20:
>Possible Causes: Wi-Fi and Atmosphere Interaction: There's a strong indication that the combination of Atmosphere and Wi-Fi usage is causing the excessive power consumption. Sleep Mode Bugs: It's possible that certain background processes in Atmosphere, related to waking up or going to sleep, are not functioning correctly, leading to the drain. EmuMMC: Running Atmosphere within emuMMC (emulated NAND) might be a factor in triggering the battery drain.

All these possible causes are not it since I've had the battery drain and it disappeared after creating a new EmuNAND and reinstalling the exact same modules.

samfadel, 2025-06-20:
> All these possible causes are not it since I've had the battery drain and it disappeared after creating a new EmuNAND and reinstalling the exact same modules.

Did u test the last firmware 20.1.5?

Fredhoob, 2025-06-20:
> Did u test the lastest firmware 20.1.5?

No, I will be waiting for an officially compatible atmosphere version to do that. I'm also worried it will cause the battery drain issue again so I'll also be waiting for reports that it doesn't once you have created a clean EmuNAND or deleted and remade your users.

impeeza, 2025-06-20:
> No, I will be waiting for an officially compatible atmosphere version to do that. I'm also worried it will cause the battery drain issue again so I'll also be waiting for reports that it doesn't [...]

Atmosphère 1.9.1  already support FW 20.1.5

Fredhoob, 2025-06-20:
I know, that's why I said "officially" so when its compatibility is mentioned in the next atmosphere release :).

SciresM, 2025-06-20:
I am not especially planning on doing a release soon, since 20.1.5 works fine.

DavidB81, 2025-06-20:
> Récemment, j'ai mis à jour ma Nintendo Switch V2 sur le firmware 20.1.1 à partir de 19.0.1.
> 
> J'ai remarqué que ma batterie se déchargeait très rapidement en mode veille (wifi activé), mais lorsque je la mettais en mode avion, il n'y avait pas de décharge de batterie. J'ai installé Atmosphere 1.9.1 et plus tard Hekate 6.3.1. Tout va bien avec les jeux. Causes possibles : Interaction Wi-Fi et atmosphère : Il y a de fortes indications que la combinaison de l'utilisation d'Atmosphere et du Wi-Fi est à l'origine d'une consommation d'énergie excessive. Bugs du mode veille : Il est possible que certains processus d'arrière-plan dans Atmosphere, liés au réveil ou à l'endormissement, ne fonctionnent pas correctement, ce qui entraîne une perte d'énergie. EmuMMC : L'exécution d'Atmosphere dans emuMMC (NAND émulée) peut être un facteur déclenchant l'épuisement de la batterie.
> 
> Solution rapide :
> 
> Désactiver le Wi-Fi

Same problem in v20.1.1 on my Mariko switch.
The CPU 3 is always at 33% charge on the home screen and the battery drains quickly. The temperature is 32°C.
Switching to airplane mode is not a solution, and I don't feel like doing the Emunand again as I have a lot of games installed.

mav2010, 2025-06-20:
Same issue here. v1 switch, Firmware 20.1.1 on sysnand. I even tried a completely freshly formatted SD card with only Atmosphere on it. Not even hekate was used. Battery drains as soon as wifi is connected.

Is there a way to do some debug logging to further look into this?

MasonT8198, 2025-06-20:
I will note on Erista the power consumption on 20.1.5 seems to be quite a bit less connected to WiFi with all updated sysmodules, I didn't run a Battery Discharge Graph on the 20.1.0 fw unfortunately, I should've thought about it, I cba to backdate fw though

samfadel, 2025-06-20:
Hi, I updated my sysnand to 20.1.5 and enabled wifi then I put it in sleep mode for 3 hrs only 1% drop (from 100% to 99%)

samfadel, 2025-06-23:
> I updated my sysnand to 20.1.5 as well, but I still have the accelerated battery drain. I am not using emunand.

I have tested both sysnand and emunand no battery drain

Alvaro-nepo, 2025-06-27:
Same issue, i will try a new emunand but i dont want to lose all so it will be a waste of time d:

Hedges, 2025-06-30:
That said, I think not all of the services listed need to be disabled. I might spend some time narrowing down which ones actually make a difference.

Hedges, 2025-06-30:
For me, it looks like `[ns.notification]` might be the main culprit - after disabling it, the battery drain issue seems to have stopped.

holictsubasa, 2025-07-01:
> [@Hedges](https://github.com/Hedges) [@holictsubasa](https://github.com/holictsubasa) These settings are not available in the official system_settings.ini of Atmosphére. The questions I have are: Are these settings supported by the official AMS and if so, which ones and where do they need to be added in the system_settings.ini? Do these settings have a negative effect on AMS and what does [@SciresM](https://github.com/SciresM) think about this?

They are supportes as they are readed from atmosphere, if they wasn't supported, then they shouldn't do anything. They just doesn't are needed to modify by default to make atmosphere works. I really don't know if the services are necessary to something, but nothing happend to my switch after disabling them. To make them work, just put them at the end of the file.

lowburger, 2025-07-02:
> For me, it looks like `[ns.notification]` might be the main culprit - after disabling it, the battery drain issue seems to have stopped.

Need ALL setting or only some? Thx

Alvaro-nepo, 2025-07-02:
okey update, after a clean install, the problem is now only with joycons on the console, try it if u want D:

Hedges, 2025-07-02:
> > For me, it looks like `[ns.notification]` might be the main culprit - after disabling it, the battery drain issue seems to have stopped.
> 
> Need ALL ns.notification setting or only some? Thx

@lowburger Didn't try disabling them individually, so can't say which one might be the cause.

Alvaro-nepo, 2025-07-02:
i did nothing about that, i will try other day killing services around joycons on sleep mode, thats my problem right now, try with no joycons and see if consume is lower than before

Tiefkuehlpizze, 2025-07-02:
> For me, it looks like `[ns.notification]` might be the main culprit - after disabling it, the battery drain issue seems to have stopped.

Just a wild thought but the issues started with firmware _20.0.0_. I don't know how you can find the names for the variables used in `system_settings.ini` but the names in `[ns.notification]` sound like the new virtual gamecard feature could fit in there. I could imagine, that the firmware has some background tasks to monitor whether there are new gamecard or a gamecard has expired. And due to the firmware modifications it's now in some kind of endless loop.
Is there a way how to find these variables?

SciresM, 2025-07-02:
These issues started in 17.0.0 or 18.0.0, actually.

holictsubasa, 2025-07-02:
> These issues started in 17.0.0 or 18.0.0, actually.

I have to disagree with this. I was on 19.0.1, I didn't have the problem. Then updated to 20.0.1 and had this problem. Then I downgraded my switch (I had to do a semi-factory reset, but that's another thing) and the problem dissapear. Then updated to 20.1.5 and the problem returned. So at least for me, the problem was only on 20.x.x

holictsubasa, 2025-07-02:
> There's nothing to disagree with. It is a matter of fact.
> 
> It began happening in 18.0.0, CTCaer and I have been receiving reports about it since then. 19.0.0 changed what consoles it happens to, and so did 20.0.0, but the fundamental issue was introduced well before then.

Oh, ok. Sorry if it looks like I'm trying to impose something. I am just trying to share my experience, since as I said, for me, it only happens on 20.x.x. Maybe that could help, it really seems that at least it increased since 20.0.0. If it doesn't help, sorry again, I only want to help.

Alvaro-nepo, 2025-07-02:
i have that problem since a lot time ago is just random shit trust me hahaha

Its-my-right, 2025-07-03:
Same here, did my first emunand on v16.0, did all the update until 20.1.5 and never noticed strange battery drain, until updating to 20.0.0.
Anyway recreating an emunand from 20.1.0 did the trick. Still suspecting some strange internet connexion loop that can not be acquitted on emunand with dans blocking

Manyone1, 2025-07-05:
it could be helpful, if someone could explain the switches (for what they are and under what circumstances they have an impact) and there meanings. Also be helpful could be the default values for these switches and differences between HOS v19 and v20 to get closer to a possible solution.

amurgshere, 2025-07-09:
> Just a wild thought but the issues started with firmware _20.0.0_. I don't know how you can find the names for the variables used in `system_settings.ini` but the names in `[ns.notification]` sound [...]

Can’t we search the code in Atmosphere to find the available options, will have a look tonight.

amurgshere, 2025-07-09:
Good news, the below fixed it for me, went from 100% to 98% in about 7 hours of stand by, all I did was add these lines to the "/atmosphere/config/system_settings.ini" and rebooted:

[ns.notification]
enable_download_task_list = u8!0x0
enable_download_ticket = u8!0x0
enable_network_update = u8!0x0
enable_random_wait = u8!0x0
enable_request_on_cold_boot = u8!0x0
enable_send_rights_usage_status_request = u8!0x0
enable_sync_elicense_request = u8!0x0
enable_version_list = u8!0x0
retry_interval_min = u32!0x7FFFFFFF
retry_interval_max = u32!0x7FFFFFFF
version_list_waiting_limit_bias = u32!0x7FFFFFFF
version_list_waiting_limit_min = u32!0x7FFFFFFF

https://github.com/hanai3Bi/Switch-OC-Suite/blob/master/system_settings.md?rgh-link-date=2025-07-09T11%3A28%3A59.000Z

I am using - Switch OLED 20.1.5 emuMMC -> 20.1.5|AMS 1.9.1|E

I know there were some instructions around for non-jailbroken switches to disable "Notifications -> Platinum Point Notification Settings", but this didn't stop the drain for me, only the above did. I also noted that once I made this change, the WiFi would disconnect when the dvice went into stand-by and reconnect as soon as you wake the device.

Hope this information helps solve the problem overall...

Fredhoob, 2025-07-17:
> Good news, the below fixed it for me, went from 100% to 98% in about 7 hours of stand by, all I did was add these lines to the "/atmosphere/config/system_settings.ini" and rebooted: [...]

So you can just add those lines to a normal atmosphere install with nothing else added and it will work? I don't have the battery drain issue at the moment but I'm saving this kind of solution as a back up if it reappears when I do another firmware upgrade.

Fredhoob, 2025-07-19:
Did anybody who fixed the abnormal battery drain via rebuilding their NAND or creating a new user have the issue return after upgrading their firmware? 

Or is it likely to be gone for good once you've fixed it?

illando, 2025-08-02:
I also fixed the problem rebuilding NAND on 20.1.5, then I recently updated to 20.3.0 and the battery drain did not return. 
I have a V1 Switch too.

EveningBlunt, 2026-10-03:
> Add the following content to **atmosphere/system_settings.ini** in the root directory of the SD card to fix the issue of abnormal standby battery drain when Wi-Fi is enabled.
> 
> ```
> [ns.notification]
> enable_download_task_list = u8!0x0
> enable_download_ticket = u8!0x0
> enable_network_update = u8!0x0
> enable_random_wait = u8!0x0
> enable_request_on_cold_boot = u8!0x0
> enable_send_rights_usage_status_request = u8!0x0
> enable_sync_elicense_request = u8!0x0
> enable_version_list = u8!0x0
> retry_interval_min = u32!0x7FFFFFFF
> retry_interval_max = u32!0x7FFFFFFF
> version_list_waiting_limit_bias = u32!0x7FFFFFFF
> version_list_waiting_limit_min = u32!0x7FFFFFFF
> ```

Random, but worked for me. Such a good feeling to find a fix. Thanks.

fudio101, 2026-11-07:
Data point on **23.0.1**, in case it helps others: the `[ns.notification]` block alone did **not** fix it for me, but adding `[bgtc]` + `[npns]` did.

**Setup:** Switch OLED (Mariko), HOS 23.0.1, Atmosphère 1.12.1, Hekate 6.5.4, partition-based emuMMC that was updated with Daybreak, Nintendo domains blocked through DNS-MITM hosts, blank PRODINFO. Same pattern as described above.

**Symptom:** about 6–8 %/hour in sleep with Wi-Fi on. Airplane Mode in sleep: about 2 % over a whole night. The drain was the same on a card with no extra sysmodules, so in my case it was not caused by homebrew.

**What I tried, each followed by about 1 hour of sleep with Wi-Fi on:**

1. `[ns.notification]` block only (the one posted above): 43 % → 37 % in 47 minutes. No change.
2. Kept that block and added these two sections (from Switch-OC-Suite's `system_settings.md`): 32 % → 32 % after about 40 minutes of sleep. The drain was gone.

```ini
[bgtc]
enable_halfawake = u32!0x0
minimum_interval_normal = u32!0x7FFFFFFF
minimum_interval_save = u32!0x7FFFFFFF
battery_threshold_save = u32!0x64
battery_threshold_stop = u32!0x64

[npns]
background_processing = u8!0x0
sleep_periodic_interval = u32!0x7FFFFFFF
sleep_processing_timeout = u32!0x0
sleep_max_try_count = u32!0x0
```

File: `atmosphere/config/system_settings.ini`. A reboot is needed after editing it.

I haven't isolated whether `[bgtc]` or `[npns]` does the work. My guess is `enable_halfawake = 0`, which stops the console from waking for background tasks in sleep. As noted above, this disables background tasks while asleep, so only use it if you don't rely on online services.

jgutierrezre, 2026-11-12:
Can confirm that @fudio101 's solution fixed the issue for me.