Atmosphere-NX/Atmosphere #2529: Error code: 2162-0002. Program: 0100000000000016

aavalenzuelap, 2025-05-12:
## Bug Report

### What's the issue you encountered?

I'm getting the error code: 2162-0002 with program: 0100000000000016. </br>
I first got this error after putting my console into sleep mode. Since then, whenever I try to boot into emuMMC or stock, the screen gets stuck on the Switch logo and the error code pops up. However, out of many attempts trying to turn the console on, I’ve only passed the boot screen twice. When it does boot, everything works just fine, but as soon as I put the console to sleep, the error returns.</br>

### How can the issue be reproduced?

1. Put the console in sleep mode, or...</br>
2. Turn it on. No matter if it's in stock or emuMMC.</br>

### Crash Report

These are the last three crashes for the same error:
1. [01747068953_0100000000000016.log](https://gist.github.com/aavalenzuelap/87baf4f70db3420373e853cffa7a5020)</br>
2. [01747054600_0100000000000014.log](https://gist.github.com/aavalenzuelap/80976678b615aa3916ca6258bd361f3c)</br>
3. [01747054556_010000000000000f.log](https://gist.github.com/aavalenzuelap/5824aaebe1b7dd868aade7b73f9a9378)</br>

### System Firmware Version

18.1.0</br>

### Environment?

Hekate v6.2.0
EmuMMC AND SysNAND:
  - EmuMMC is partition-based

### Additional context?

Important information:
  - I haven't downloaded anything new in the last couple of weeks
  - In my opinion, the Wi-Fi IC is still working properly, so I maybe this error is not related to it. I replaced it a year ago and since then, I've never used the internet.

[image]

SciresM, 2025-05-12:
I am pretty firmly convinced that this is not an atmosphere issue, but something going on with your hardware. Going to close this issue, but yeah. Fwiw, this issue tracker is for atmosphere bugs, not for support, which is given on discord.

closed 2025-05-12 by SciresM

kil29060-ctrl, 2026-09-27:
Absolutely the same problem. What to do? Absolutely the same problem. What to do?

aavalenzuelap, 2026-09-27:
It turned out to be a hardware issue. The IC chip M92T36 had stopped working properly. Once replaced the switch went back to normal. Hopefully the solution is the same as mine.