Atmosphere-NX/Atmosphere #2533: Option to completely disable WiFi hardware module

nilldot, 2025-05-16:
It’s not about the fear of ban, or anything. The wifi chip is physically damaged, resulting in annoying Error [2110-2118](tel:2110-2118) message after every game change in Mariko switch. Oddly enough, switch can see wireless SSIDs, but can’t connect, including manual connections. Everything else is working as expected. I checked the settings.ini file, but there is nothing indicating options to disable the said. I wonder if it would be possible to suggest/introduce an option to completely disable hardware checks during the boot on the homebrew level?

Thank you

borntohonk, 2025-05-17:
> It’s not about the fear of ban, or anything. The wifi chip is physically damaged, resulting in annoying Error 2110-2118 message after every game change in Mariko switch. Oddly enough, switch can see wireless SSIDs, but can’t connect, including manual connections. Everything else is working as expected. I checked the settings.ini file, but there is nothing indicating options to disable the said. I wonder if it would be possible to suggest/introduce an option to completely disable hardware checks during the boot on the homebrew level?
> 
> Thank you

Is this a botched modchip install situation? Either way if it is or isn't, I'm pretty sure i recall the wifi chip also does the Bluetooth communication.

my first thing to try would just be to connect an ethernet adapter into your dock, and if it no longer errors, then there's your fix.

as for disabling the wlan services, i think that should naturally occur if you use the ethernet services.

disabling wlan services in abnormal fashion, i think nifm (and maybe LDN?) wouldn't like.

nilldot, 2025-05-17:
Thanks. This is indeed botched modchip install situation, which is too expensive to fix at this stage. The thing is, with Tinwoo and DBI in place, I truly don’t need wifi. Yes, it’s annoying, and I understand nuances with  FTP transfers( slow btw), I can leave without one. Unfortunately, that message is truly annoying for every occasion when changing the game. Perhaps, the other tix is to disable notifications all alone, opposite to hardware checks, etc.

Thank you

[image]

impeeza, 2025-05-17:
> Thanks. This is indeed botched modchip install situation, which is too expensive to fix at this stage. The thing is, with Tinwoo and DBI in place, I truly don’t need wifi. Yes, it’s annoying, and I understand nuances with FTP transfers ( slow btw), I can leave without one. Unfortunately, that message is truly annoying for every occasion when changing games. Perhaps, the other tix is to disable notifications all alone, opposite to hardware checks, etc.
> 
> Thank you
> 
> [image]

There are solutions already in place, you should ask on a support site like GBATemp. as this is not a Atmosphère issue please close this and ask on GBATemp.

zemixe, 2025-05-17:
Piracy is not supported.

nilldot, 2025-05-17:
I’m probably missing something here. I’m not asking for mod assistance, games, or anything. I’m asking for support on atmosphere functionality, nothing more, nothing less. Not sure what’s the twist is all about, really. If option to disable wifi is not available, perhaps there is a mean to completely disable all sort and possibly alerts, which would include said notification on the firmware level.
Thanks

SciresM, 2025-05-17:
1) piracy is not supported
2) I do not see any reason for this to be part of atmosphere as opposed to e.g. some bespoke custom system module or patch to wlan.

closed 2025-05-17 by SciresM