Atmosphere-NX/Atmosphere #2535: Need Some Help with Error Code: 2110-0350 (0x2bc6e)

SDM855, 2025-05-17:
## Bug Report
### What's the issue you encountered?
[image]
I am running a custom bot script that uses sys-botbase to read game data(Pokémon SWSH) from a specific pointer (works similarly to PKHeX-LiveHeX / PyNXBot / PyNXReader).

This bot works forever if I don't restart the game. However, for specific hunting, I need to close and restart the game: in this case, if I run the script and restart the game around 20 times(about 30 minutes), the Atmosphere will crash for Error Code: 2110-0350 (0x2bc6e), and I can repeat this crash every attempt.

I tried it on 18.1.0 / 20.0.1, same problem; I tried it on a fresh new emuMMC(FAT32, SD partition), same problem.
This code is so rare that I have no clue about it. The only thing I found is
https://github.com/Atmosphere-NX/Atmosphere/issues/1984
It was related to PKHeX-LiveHeX / sys-botbase and was defined as not being an atmosphere issue.

### How can the issue be reproduced?

As mentioned above, Atmosphere will crash if I use the script and restart the game around 20 times(about 30 minutes).

### Crash Report
https://gist.github.com/SDM855/f9d71f3663cf2408ea08d158635926fa
https://gist.github.com/SDM855/deafc389959d6753a26ae63539d31f34
https://gist.github.com/SDM855/82b15d51c027ec993992c405b66bb618

### System Firmware Version

18.1.0 (Atmosphère 1.9.0)
20.0.1 (Atmosphère 1.9.0)

### Environment?

- hekate: 6.2.5 /6.3.0
- AK478BB: atmosphere-1.9.0-20_support-9dd8269f7-dirty
- N/A
- Homebrew software: just some basic ones
- EmuMMC
- FAT32 SD Partition

### Additional context?
I am not asking for a fix because it is a rare issue that might not relate to the Atmosphere. I would appreciate it if someone could provide me with more information about the error or if there is any way to avoid it. Thank you so much!

Lusamine, 2025-05-18:
> AK478BB: atmosphere-1.9.0-20_support-9dd8269f7-dirty
> Homebrew software: just some basic ones

Normally, it would help to actually list what you have running if you are asking for help. I looked up your Atmosphère install and it looks like it bundles a ton of stuff...

Your issue is occurring because of ldn_mitm, which has this problem when resetting SWSH. I am able to reproduce the crash by resetting SWSH 19 times manually with ldn_mitm rebuilt with latest Atmosphere-libs. This has been the case since sometime around late 2023.

[image]

SDM855, 2025-05-18:
> Your issue is occurring because of ldn_mitm, which has this problem when resetting SWSH. I am able to reproduce the crash by resetting SWSH 19 times manually with ldn_mitm rebuilt with latest Atmosphere-libs. It crashes when hitting the overworld on the 20th reset. This has been the case since sometime around late 2023.
> 
I appreciate your help. Yes, I did use ldn_mitm (spacemeowx2 version and DefenderOfHyrule version), and the same problem.

Does this problem only happen to SWSH? Is there any solution to this? Thanks

Lusamine, 2025-05-18:
I saw that you edited your original post to list your homebrew software. I feel the only thing I can tell you now is that piracy is not supported.

borntohonk, 2025-05-18:
> I appreciate your help. Yes, I did use ldn_mitm (spacemeowx2 version and DefenderOfHyrule version), and the same problem.
>
> Does this problem only happen to SWSH? Is there any solution to this? Thanks

The problem is caused by ldn_mitm, file issue in their github tracker. Not atmosphere issue, as pointed out above.

closed 2025-05-18 by SDM855