Atmosphere-NX/Atmosphere #2538: Not working touchscreen

cloge01, 2025-05-18:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

Touchscreen works perfect on HEKATE but Nothing in SYSNAND/EMUNAND

### How can the issue be reproduced?
i upgrade hos 20.0.1 then I chipped my nsw lite and download ultranx and this happens , i tried unchip , load fullstock, i checked everything inside switch and in switch all fine, tried replase sd card, tried to replace card slot and it not work.

### System Firmware Version
 
20.0.1

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
- Official release or unofficial build: ultraNX new version
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading: -
- Homebrew software installed: [ - ]
- EmuMMC or SysNAND: EmuMMC
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

i checked all forums and some people had this problem but not have solution

cloge01, 2025-05-18:
[image]

[image]

borntohonk, 2025-05-18:
> [image]
> 
> [image]

Your title is misleading, as it says "Touchscreen works perfect on HEKATE"

yet your picture says "touchscreen calibration failed"

anyway sounds like you've messed up modchip installation/reassembly of console.

not really an atmosphere problem

> download ultranx
then contact them for support, why even open an issue in the atmosphere issue tracker then?

cloge01, 2025-05-18:
anyway sounds like you've messed up modchip installation/reassembly of console.
ultranx use atmosphere as cfw so i just open issue in atmo ultra and hekate just warned maybe you can fix or smth in next update bcos i think it not hardware issue

but if i messed up touch will not work in hekate but in work.

Konzolista, 2025-05-18:
> tak znie, ako by ste chaos modchip inštalácia/nové zostavenie konzoly. ultranx použiť atmosféru ako cfw tak som stačí otvoriť problém v atmo ultra a hekate len varoval, možno môžete opraviť alebo smth v ďalšej aktualizácii bcos myslím, že to nie je problém hardvéru
> 
> ale ak som skazil dotyk nebude pracovať v hekate, ale je to práca.

If you can't even touch sysNAND OFW, it's not an AMS problem, OMG !!!

Chimecho, 2025-05-24:
This happened to me once but with an OLED and I couldn't find the cause, still haunts me to this day. Only thing I could think of after a long time is that maybe some flux got under the touch controller or smth, but couldn't confirm since the console wasn't with me anymore.

Anyways, try to give it an ultrasonic bath and maaaybe it will work again.

souldbminersmwc, 2025-05-24:
> anyway sounds like you've messed up modchip installation/reassembly of console. ultranx use atmosphere as cfw so i just open issue in atmo ultra and hekate just warned maybe you can fix or smth in next update bcos i think it not hardware issue
> 
> but if i messed up touch will not work in hekate but it work.

Piracy is not supported