Atmosphere-NX/Atmosphere #2539: Disable enabling of cheats by default

souldbminersmwc, 2025-05-19:
## Feature Request

### What feature are you suggesting?
#### Overview:
Disabling the enabling of all cheats upon launch

#### Smaller Details: 
Change default config value

#### Nature of Request:  
[ Remove all that do not apply to your request. ]
- Change
- Config value enabling cheats

### What component do you feel this would best fit within?

- [**Troposphère**](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Application-level patches to the Horizon OS.</br>

### Why would this feature be useful?
It helps by not having to toggle off cheats for people who are new to homebrew, and it can generally be an annoyance, as certian cheats, like DRS or version-specific cheats, can conflict and breaj things

SciresM, 2025-05-19:
No, cheats on by default is preferable behavior. Users are free to not install cheats they don't want on by default, or change the cheat enable key on a game by game basis.

SciresM, 2025-05-19:
You should never have "version specific cheats" breaking things btw, since cheats are loaded based on the build ID, which is version specific. Very illogical argument.

closed 2025-05-19 by SciresM

SciresM, 2025-05-20:
Sounds like a problem with your cheats, not with atmosphere. I recommend using greater discretion in what cheats you're using.

Anyway, locking this, since it's resolved.