Atmosphere-NX/Atmosphere #2542: Atmosphere 1.9.0 causes some games not starting

oujiafan, 2025-05-25:
## Bug Report

N/A

### What's the issue you encountered?

After installing Atmosphere 1.9.0, some games cannot start (some are fine though). Tinfoil cannot start either.
It will prompt "unable to start software".

If downgrading to Atmosphere 1.8.0, tinfoil and other games will be back to noraml.

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

N/A

### System Firmware Version

18.1.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Fusee
- Official release or unofficial build: 
  - Fusee released with AMS 1.9.0

- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

FourHawk, 2025-05-25:
Piracy is not supported

borntohonk, 2025-05-25:
> Tinfoil cannot start either.
> It will prompt "unable to start software".

Piracy is not supported

closed 2025-05-28 by SciresM