Atmosphere-NX/Atmosphere #2545: 20.1.0 needs comparability

JCM-devv2, 2025-05-28:
Well I updated my switch lite sysnand to 20.1.0 no longer boots gives me a fatal error I will post a picture of the issue 
But it looks like it needs another patch for compatibility 

[image]

ToxShade, 2025-05-28:
I wonder how long it will take for the full version of 1.9.0 to be released soon

prf8814, 2025-05-28:
Any release date info?

kirakira-commits, 2025-05-28:
Welp. nintendo is definitely working a lot on updates lately

Arnold101knight, 2025-05-28:
> > > > Any release date info?
> > > 
> > > 
> > > Yesterday
> > 
> > 
> > clown
> 
> The only clown is you. There's never been a release date. Why don't you do it yourself if you're in such a hurry? If you did more research you'd know that sciresM was really sick and still needs to recover. And even he can't estimate how long it will take. It's ready when it's ready. It's your own fault if you update immediately every time. You should hammer it into your heads that in the CFW scene you never update without checking for compatibility first.

you are the clown, clown! i'm not even the guy that asked for it, CLOWN, get a lif instead of instand-reply CLOWN

SilverT4, 2025-05-28:
never a good idea to update HOS until you know there's a compatible release of atmosphère. admittedly, i did that exact thing last night on my switch, so i'm in the same boat of atmosphère not working.

DamImHereForIssues, 2025-05-28:
#/ @SilverT4 unfortunately I knew about the update was gonna brick so didn't update, that was until... I was gonna play a game and I mashed the A button so now I'm here :|

MisterCRD92, 2025-05-28:
Nobody cares about your pirated opinions. Let sciresM work in peace and go to a dirty forum.

SonicStylin91, 2025-05-30:
i’m smart enough to know that if u update your a big dumb clown like Arnold101knight

shooterspps, 2025-05-30:
Is the USB 3.0 patch in 20.1.0 the same as in 20.0.1? I can't enable USB 3.0 in DBI on Atmosphere 20.1.0, but when I switch to Atmosphere 20.0.1 or an earlier version, USB 3.0 works normally. I noticed in ldr_embedded_usb_patches that it only goes up to 20_0_0.

shooterspps, 2025-05-31:
> > Is the USB 3.0 patch in 20.1.0 the same as in 20.0.1? I can't enable USB 3.0 in DBI on Atmosphere 20.1.0, but when I switch to Atmosphere 20.0.1 or an earlier version, USB 3.0 works normally. I noticed in ldr_embedded_usb_patches that it only goes up to 20_0_0.
> 
> Piracy is not supported!

You're such a genius.

lucyamonster, 2025-05-31:
> > Is the USB 3.0 patch in 20.1.0 the same as in 20.0.1? I can't enable USB 3.0 in DBI on Atmosphere 20.1.0, but when I switch to Atmosphere 20.0.1 or an earlier version, USB 3.0 works normally. I noticed in ldr_embedded_usb_patches that it only goes up to 20_0_0.
> 
> Piracy is not supported!

There can't be another reason why they want fast file transfers. You only need to transfer files for piracy. We all know that for other things the bits just magically appear on the sd card. Only for evil piracy somebody would need a cable /s

dxgldotorg, 2025-05-31:
> There can't be another reason why they want fast file transfers. You only need to transfer files for piracy. We all know that for other things the bits just magically appear on the sd card. Only for [...]

Maybe it only breaks that pirate tool?

Visualwit, 2025-05-31:
How is this issue still open? The 1.9.1 update is out now with HOS 20.1.0 support!

newdev8, 2025-06-01:
@Visualwit I get a black screen when trying to boot Atmosphere 1.9.1 with HOS 20.1.0...

dxgldotorg, 2025-06-01:
> [@Visualwit](https://github.com/Visualwit) I get a black screen when trying to boot Atmosphere 1.9.1 with HOS 20.1.0...

If you are getting the panic screen and you have Mission Control, update that on your microSD.

newdev8, 2025-06-01:
Not even a panic screen, I hit EmuMMC and it goes to a black screen, forever, until I reboot.

newdev8, 2025-06-02:
Thank you! That fixed it!

JCM-devv2, 2025-06-03:
Welp there's another update any possibility this one breaks atmosphere

SonicStylin91, 2025-06-03:
i didn’t update my Switch did you?

SonicStylin91, 2025-06-03:
do you mean i helped?