Atmosphere-NX/Atmosphere #2547: error code 2168-0002 (0x4a8)

jamesleung1234, 2025-05-29:
## Bug Report

Been having this issue since the atmosphere update. Saw a few posts about this issue but none can suit in my situation. I hope someone can help me to solve this</br>

### What's the issue you encountered?

When I open sysnand it will crash </br>

### Crash Report

Quite a lot of crash report related to this

[01746873243_010000000000000f.log](https://github.com/user-attachments/files/20504836/01746873243_010000000000000f.log)
[01746873287_010000000000000f.log](https://github.com/user-attachments/files/20504841/01746873287_010000000000000f.log)
[01746873429_010000000000000f.log](https://github.com/user-attachments/files/20504840/01746873429_010000000000000f.log)
[01746873463_010000000000000f.log](https://github.com/user-attachments/files/20504837/01746873463_010000000000000f.log)
[01747034681_010000000000000f.log](https://github.com/user-attachments/files/20504838/01747034681_010000000000000f.log)
[01747038342_010000000000000f.log](https://github.com/user-attachments/files/20504842/01747038342_010000000000000f.log)
[01748522477_010000000000000f.log](https://github.com/user-attachments/files/20504839/01748522477_010000000000000f.log)

</br>

### System Firmware Version

20.0.1</br>

borntohonk, 2025-05-30:
> ## Bug Report
> 
> Been having this issue since the atmosphere update. Saw a few posts about this issue but none can suit in my situation. I hope someone can help me to solve this
> ### What's the issue you encountered?
> 
> When I open sysnand it will crash
> ### Crash Report
> 
> Quite a lot of crash report related to this
> 
> [01746873243_010000000000000f.log](https://github.com/user-attachments/files/20504836/01746873243_010000000000000f.log) [01746873287_010000000000000f.log](https://github.com/user-attachments/files/20504841/01746873287_010000000000000f.log) [01746873429_010000000000000f.log](https://github.com/user-attachments/files/20504840/01746873429_010000000000000f.log) [01746873463_010000000000000f.log](https://github.com/user-attachments/files/20504837/01746873463_010000000000000f.log) [01747034681_010000000000000f.log](https://github.com/user-attachments/files/20504838/01747034681_010000000000000f.log) [01747038342_010000000000000f.log](https://github.com/user-attachments/files/20504842/01747038342_010000000000000f.log) [01748522477_010000000000000f.log](https://github.com/user-attachments/files/20504839/01748522477_010000000000000f.log)
> 
> ### System Firmware Version
> 
> 20.0.1

Remove sys-patch

piracy is not supported

WibWobYT, 2025-05-30:
> Remove sys-patch
>
> piracy is not supported

This means that it is impossible to use atmosphere on the switch and this for all the people who have updated the firmware?

Masamune3210, 2025-05-30:
Nobody said that?

jamesleung1234, 2025-05-30:
> Fill out the rest of the template. It's not some checklist you get to ignore parts of just because you don't want to.

sorry about that. Updated it

jamesleung1234, 2025-05-30:
> Remove sys-patch
>
> piracy is not supported

even though i remove syspatch it still crash

WibWobYT, 2025-05-30:
I found a solution, I updated the sys-patches that are compatible with 20.0 and it worked again for me. Try it but don't forget to backup the files beforehand in case it doesn't work.

Masamune3210, 2025-05-30:
What part of "Piracy is not supported" made you think it would be ok to continue talking about it?

antonwe, 2025-05-30:
> What part of "Piracy is not supported" made you think it would be ok to continue talking about it?

probably the part where a piracy-associated sysmodule mitigates a non-piracy-related atmosphere crash.

SciresM, 2025-05-30:
This is piracy, not supported.

closed 2025-05-30 by SciresM