Atmosphere-NX/Atmosphere #2549: Error when booting into Atmosphere 0x4A8 (2168-0002)

Idain, 2025-05-30:
## Bug Report

### What's the issue you encountered?

I tried to update Atmosphere to v1.9.1 along with Hekate to v6.3.1, and finally update my Switch firmware to v20.1.0 from my sysCFW. When rebooting the console and trying to access sysCFW, it shows the Switch logo for a few seconds and then shows an error screen with the error code 2168-0002 (0x4A8).

### How can the issue be reproduced?

1. Update Atmosphere to v1.9.1
2. Update Switch firmware to v20.1.0
3. Reboot the console
4. Access sysCFW from Hekate

### Crash Report

Link: https://gist.github.com/Idain/fa5d8b750b8af4c2c49d3f0fb7f47ec0

### System Firmware Version

20.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: 
  - Hekate v6.3.1

- Official release or unofficial build: 
  - Official release v1.9.1
  
- Do you have additional kips or sysmodules you're loading: 
  - sys-con
  - MissionControl
  - TeslaMenu
  - nx-ovlloader
  - sys-patch
 
- Homebrew software installed: 
  - Goldleaf
  - JKSV
  - FTPD
  - NXTheme Installer
  - NX-Shell
  - Daybreak
  - Homebrew Store
  - SimpleModManager
  - Awoo-Installer
  - DBI
  - EdiZon

- EmuMMC or SysNAND:
  - SysNAND

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

IHadK, 2025-05-30:
0100000000001000 is a custom theme, delete 0100000000001000 from your contents folder under atmosphere and then try to boot again.

Idain, 2025-05-30:
Oh wow, that solved the problem! Thank you!

closed 2025-05-30 by Idain