Atmosphere-NX/Atmosphere #2554: hbmenu doesn't go in

mbtipass, 2025-06-02:
## bug report

### What kind of problem did you have?

### How can I recreate the problem?

Renamed and ran the Hekate file as PAYLOAD at the top of SDCARD...
Hekate got in, I got into SYSCFW, but hbmenu doesn't go in. Only the album screen keeps showing up
I searched for the solution and changed the contents of the ini file, but the error is not fixed
But you can enter to syscfw, but the problem is that you can't enter hbmenu 

### System Firmware Version

I downloaded ofw 20.1.0 

### Environment?

- What bootloaders (fusee, hekate, etc.) were released by Atmospère: hekate v6.3.1 & Nyx v1.7.0
- Homebrew Software Installation: goldleaf, jksv, nx-shell, nxthemeinstaller, ftpd-classic
- EmuMMC , SysNAND
  - If you use EmuMMC, include partition-based or file-based -> partition-based

Please help me with this... I have no clue

mbtipass, 2025-06-02:
I copied the file as is and put it in the root of the sd card. But it still doesn't turn on. It doesn't work even if I press R and click on the album.

mbtipass, 2025-06-02:
> Do you have the hbmenu.nro file in your root directory on your SD card? If not, I'd add it; otherwise, it won't load. Of course, you also need the hbloader, but that's all included with the Atmosphere release.

> SD 카드의 루트 디렉토리에 hbmenu.nro 파일이 있습니까? 그렇지 않다면, 나는 그것을 추가할 것이다; 그렇지 않으면, 그것은 로드되지 않을 것이다. 물론 hbloader도 필요하지만, 그것은 모두 대기 방출에 포함됩니다.

mbtipass, 2025-06-02:
> ## 버그 리포트
> ### 어떤 문제가 있었나요?
> ### 어떻게 하면 문제를 재현할 수 있을까요?
> 스위치 가이드에 따라 모든 SD 파일을 설정 하고 SDCARD 상단에 PAYLOAD로 Hekate 6.3.1 bin 파일을 실행했습니다. Hekate가 들어가서 SYSCFW에 들어갔지만 hbmenu가 들어가지 않습니다. 앨범 화면만 계속 표시됩니다 . "more configs" 메뉴에는 "No .ini 또는 boot enries가 표시됩니다. .ini 파일이 부트로더/ini/에 존재하고 적어도 하나의 항목이 포함되어 있는지 확인합니다. 해커테_ipl.ini 파일을 부트로더 폴더에 넣을 준비를 하고 솔루션을 검색하고 ini 파일의 내용을 올바르게 변경했지만 오류가 고정되지 않았습니다 . syscfw에 입력할 수 있으며 현재 버전이 20.1.0 l AMS l S인 것을 확인했지만 hbmenu에 입력할 수 없는 것이 문제입니다.
> 
> ### 시스템 펌웨어 버전
> 20.1.0w 다운로드 받았습니다
> 
> ### 환경?
> * 아트모스페르가 출시한 부트로더(퓨지, 헤카테 등)는 아트모스페르 1.9.1, 헤카테 v6.3.1 & Nyx v1.7.0
> * Homebrew Software 설치: Goldleaf, jksv, nx-shell, nxthemeinstaller, ftpd-classic
> * 에뮤MMC, 시스낸드
>   
>   * EmuMMC를 사용하는 경우 파티션 기반 또는 파일 기반 -> 파티션 기반을 포함합니다.
> 
> 이거 좀 도와주세요... 전혀 모르겠어요.

[image]
[image]
[image]
[image]

Aura67, 2025-06-02:
If you want to open the hbmenu via the album, you don't have to press the R key. You just have to open your album with the A key. If you hold down R and then press A on the album, the standard directory where your screenshots go will open. It's different if you want to open the hbmenu via a game. You have to hold down the R key and start a game, then the hbmenu should open in full RAM.

mbtipass, 2025-06-02:
When I use Kefir, I open it by pressing R + album touch, but it's not even in applet mode. When I use only Wait and Hecate, do I just press A? And leave out R altogether...?

Aura67, 2025-06-02:
the normal settings are in atmosphere, if you want to open the hbmenu via the album the A button only without R but if you want to start the hbmenu via a game then you have to hold down R and then start a game then the hbmenu should open in full ram, but if you press R on the album as you described and then press A on the album then the album will open but without the hbmenu then you will only see the screenshots you took so you could say it is the normal album from an ofw from the stock firmware without homebrew.

mbtipass, 2025-06-02:
Thank you so much teacher! I just pressed a and homebrew started, but it says applet mode. How do I get into normal mode?

mbtipass, 2025-06-02:
> Thank you so much teacher! I just pressed a and homebrew started, but it says applet mode. How do I get into normal mode instead of this mode?

mbtipass, 2025-06-02:
I installed the game and tried playing it, but the normal home menu only opens once out of five times. The game boots up the rest of the time. Is there a way to open the normal home menu right away even if I install something else?

mbtipass, 2025-06-02:
installed the game and tried playing it, but the normal home menu only opens once out of five times. The game boots up the rest of the time. Is there a way to open the normal home menu right away even if I install something else?

mbtipass, 2025-06-02:
> To get into normal mode (full ram) install an original game that you have bought you will not be banned if you use the online service with it you can download it normally via the nintendo service and when it is installed just hold R and start a game with the A key

installed the game and tried playing it, but the normal home menu only opens once out of five times. The game boots up the rest of the time. Is there a way to open the normal home menu right away even if I install something else?

I accidentally closed this post. Is there a way to reopen it? This is my first time on this site, so I'm making a lot of mistakes.

Aura67, 2025-06-02:
Every time you want to open the hbmenu in full ram you have to hold down R and start the game and hold R down. When you want to open the game you just need to press A, just like normal, but you want the hbmenu so hold R down while it starts.

mbtipass, 2025-06-02:
Could it be that I'm not from an English speaking country and I got the wrong message? I don't use pirated copies. I was just asking if I could remove the notification telling me to update to Goldleaf. How do I tell when an update is major?

mbtipass, 2025-06-03:
Oh, I think the booting is wrong, so I saw it while fixing it, so I don't know the deeper content. I don't know how to pirate or do it, and I don't want to do it, and thank you for helping me solve it.

Aura67, 2025-06-03:
anyway a new switch firmware is out, but its working with atmospehere 1.9.1 you  can update it. ;) and no problem your problem is not a atmospehre issues. :)

mbtipass, 2025-06-03:
Ah, yes!! I was mistaken and looking into it. After getting help, it's not an artosphere problem.

closed 2025-06-03 by mbtipass