Atmosphere-NX/Atmosphere #2557: Version 20.1.1  support

Drakul-l, 2025-06-03:
Ver. 20.1.1 (Released June 2, 2025)
Fixed an issue where some software would not start after updating to system version 20.1.0.

Not sure if it affects the existing release with support for 20.1.0.

T3ll3S-05, 2025-06-03:
I updated by accident but nothing changed, everything still works fine.

isuraaae, 2025-06-03:
I've updated it all the time, it's still trial and error but as long as I use it all functions are still normal there are no errors or significant changes.

setzer333, 2025-06-04:
We need an update of atmosphere ASAP please.

closed 2025-06-04 by SciresM

Jusabi, 2025-06-17:
My gf accidentally updated to 20.1.1 now online stopped working was working fine is it because Atmosphere doesnt support 20.1.1 yet? I hope not sure what to do.

T3ll3S-05, 2025-06-17:
> My gf accidentally updated to 20.1.1 now online stopped working was working fine is it because Atmosphere doesnt support 20.1.1 yet? I hope not sure what to do.

Pray you didn't get banned, check if you are blanking prod info in exosphere.ini

SonicStylin91, 2025-06-17:
let me know if it works for you.

Jusabi, 2025-06-17:
What you can remove a ban that's crazy I am just going to sleep just got done with work. I'm gonna try it tomorrow and I will let you know praying something works, lol.  Also I have been wanting to start fresh, like a complete restart/reset. Like how I had it from the very beginning, but I have changed sooo many random files watching sooooo many different random youtube videos, the card is just messed up, honestly it's probably got stuff on there. It don't even need or stuff interfering with other stuff, to be honest.

T3ll3S-05, 2025-06-17:
> What you can remove a ban that's crazy I am just going to sleep just got done with work. I'm gonna try it tomorrow and I will let you know praying something works, lol. Also I have been wanting to [...]

You can probably clean 90% of that with dbi

InfinityCoda, 2025-06-18:
> My gf accidentally updated to 20.1.1 now online stopped working was working fine is it because Atmosphere doesnt support 20.1.1 yet? I hope not sure what to do.

@Jusabi It's probably not a ban like suggested above. A banned console will give a specific message when trying to use online services, stating that use of online services has been restricted. And no, you can't remove a ban, bans are server side and nothing you do will clear those logs. That guy just made that up along with all that nonsense about private browsing windows hiding your IP and Nintendo suing people for logging into their own accounts. 

If you're not able to connect to wifi at all, it's more likely something with the network. If you're connected to wifi but failing to reach Nintendo services, this could be due to measures like DNS MITM / Exosphere being used to prevent connecting to Nintendo's servers. In either case it's somdthing to do with your setup rather than an atmosphere bug. The discord is a better place for this sort of thing.

SonicStylin91, 2025-06-18:
@Jusabi if you wanna get nerdy and technical don’t dude! I said my acdount banned not my console banned brotha! so if your gonna get on me CHECK first. but yea if your switch is banned and you wanna reset everything anyway u could system wipe but ima not very technical technical like @Jusabi

Jusabi, 2025-06-19:
Sorry you guys been super busy I'm gonna try a few things you all recommended I gotta figure this out because she played online ONLY with Games we owned for about a year and never got banned or anything so I gotta get it working again ha-ha. Veryyy frustrating.  Also found out the Micro SD card is corrupted and i can't take some files off to transfer to the new one I ordered also idk what i need from the old one to the new one anyways tbh