Atmosphere-NX/Atmosphere #2561: Cannot open games after updating

kirakira-commits, 2025-06-10:
## Bug Report

Cannot launch any games after updating Atmosphere to 1.9.1 and hekate to 6.3.1 and Nyx to 1.7.0. I'm on a hardmodded switch, not sure if thats the problem. (my atmosphere used to be 1.8.0) (ofw v19.0.0 china ver)

Other things are working fine, like the hbmenu, the useer page, settings, etc

All of my games were dumped from my real game cards and it worked before updating.

### What's the issue you encountered?

I did nothing peculiar, just played some games and decided to update atmosphere.

### How can the issue be reproduced?

I'm not sure. All my actions leading to the issue is listed below:

- Played games.
- Current AtmosphereVer: 1.8.0, target is v1.9.1, ofw ver is (china ver) 19.0.0
- Downloaded the latest release of atmosphere from github
- replaced the atmosphere folder and fusee.bin (didnt touch payload.bin)
- Booted it up, had an error, and then I remembered to update hekate
- Booted into emuMMC successfully, but cannot open any games. (all games, including DBI and Tinfoil. can still access and run hbmenu applets)
- Error message: Unable to launch application. please try again in the home menu

### Crash Report

no crash

### System Firmware Version

19.0.0 china ver</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - Hekate 6.3.1 and Nyx 1.7.0
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - File-based

### Additional context?

- Additional info about your environment:
- my heakte used to be in chinese, updated to english

masagrator, 2025-06-10:
Atmosphere never supported playing dumps. That's not an atmosphere issue. Any attempt at playing dumps is treated as piracy by Nintendo and by extent - Atmosphere. Your source doesn't matter

borntohonk, 2025-06-10:
Piracy is not supported.