Atmosphere-NX/Atmosphere #2562: Error booting into SYSCFW 0x4A8 (2168-0002)

CloverHouraisan, 2025-06-11:
## Bug Report

### What's the issue you encountered?

When I tried to boot into sysCFW after updating my firmware to 20.1.1 via TegraExplorer, I was shown an error screen with the code 0x4A8 (2168-0002) after I pressed the same button three times like you normally do upon launching your Switch.  I can still launch emuMMC perfectly fine and update my firmware with old methods where you have to launch your stock, load in your payload via TegraRCMGUI, and dump your firmware that way, but I'd like to have this fixed so I can update with newer methods involving sysCFW.

### How can the issue be reproduced?

1. Update Atmosphere to v1.9.1
2. Update Hekate to v6.3.1
3. Update Switch firmware to v20.1.1 via TegraExplorer
4. Reboot the console
5. Launch sysCFW from Hekate
6. When prompted, press the same button on the console 3 times to proc error

### Crash Report

Link to log: https://gist.github.com/CloverHouraisan/ef979072e740d63b79a9c247e5ee791a

### System Firmware Version

20.1.1

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
    - Hekate
- Official release or unofficial build: 
  - v6.3.1 official release
- Do you have additional kips or sysmodules you're loading:
  - None that I'm aware of
- Homebrew software installed:
  - Goldleaf
  - Daybreak
  - JKSV
  - NX Themes Installer (never used)
  - Homebrew App Store (never used)
  - Checkpoint (never used)
- EmuMMC or SysNAND:
  -emuMMC

### Additional context?

N/A

Cypherous, 2025-06-11:
Yeah it doesn't support 20.1.1 afaik and Michael is on holiday until the 15th so probably won't be fixed for a few days, there wasn't much point in updating yet

CloverHouraisan, 2025-06-11:
Okay, that's good to know.  I'll just keep my eyes open for any future updates and adjust accordingly.

borntohonk, 2025-06-11:
> Yeah it doesn't support 20.1.1 afaik and Michael is on holiday until the 15th so probably won't be fixed for a few days, there wasn't much point in updating yet

This is fake news

> ## Bug Report
> ### What's the issue you encountered?
> 
> When I tried to boot into sysCFW after updating my firmware to 20.1.1 via TegraExplorer, I was shown an error screen with the code 0x4A8 (2168-0002) after I pressed the same button three times like you normally do upon launching your Switch. I can still launch emuMMC perfectly fine and update my firmware with old methods where you have to launch your stock settings to update your sysMMC the normal way, enter RCM with a jig, load in TegraExplorer via TegraRCMGUI, and dump your firmware that way, but I'd like to have this fixed so I can update with newer methods involving sysCFW.
> ### How can the issue be reproduced?
> 
>     1. Update Atmosphere to v1.9.1
> 
>     2. Update Hekate to v6.3.1
> 
>     3. Update Switch firmware to v20.1.1 and dump with TegraExplorer
> 
>     4. Update emuMMC firmware via Daybreak
> 
>     5. Reboot the console
> 
>     6. Launch sysCFW from Hekate
> 
>     7. When prompted, press the same button on the console 3 times to proc error
> 
> 
> ### Crash Report
> 
> Link to log: https://gist.github.com/CloverHouraisan/ef979072e740d63b79a9c247e5ee791a
> ### System Firmware Version
> 
> 20.1.1
> ### Environment?
> 
>     * What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
>       
>       * Hekate
> 
>     * Official release or unofficial build:
>       
>       * v6.3.1 official release
> 
>     * Do you have additional kips or sysmodules you're loading:
>       
>       * None that I'm aware of
> 
>     * Homebrew software installed:
>       
>       * Goldleaf
>       * Daybreak
>       * JKSV
>       * NX Themes Installer (never used)
>       * Homebrew App Store (never used)
>       * Checkpoint (never used)
> 
>     * EmuMMC or SysNAND:
>       -emuMMC
> 
> 
> ### Additional context?
> 
> N/A

Delete sys-patch (SD:/atmosphere/contents/420000000000000B/)
Piracy is not supported.