Atmosphere-NX/Atmosphere #2563: Atmosphere Error Crash — 2168-0002

Naber96, 2025-06-14:
### What's the issue you encountered?
When I launch I am getting some kind of error message 

Describe the issue in detail and what you were doing beforehand.
I was trying to update fw to 20.1.1
Did you make any changes related to Atmosphère itself
Updated to 1.9.1

### How can the issue be reproduced?
Just launch the switch

### Crash Report

<script src="https://gist.github.com/Naber96/aa0131facdfffc215bf46982533d2c72.js"></script>
](https://gist.github.com/Naber96/aa0131facdfffc215bf46982533d2c72)

### Crash Report

### System Firmware Version
20.1.1

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Hekate

borntohonk, 2025-06-23:
> i am having the exact same issue but cannot find a solution please help

Your issue at https://github.com/Atmosphere-NX/Atmosphere/issues/2568 is completely unrelated to this one, so that is incorrect.

yesnrrd, 2025-08-02:
Same error with accidental update to 20.3.0 https://imgur.com/a/eu0CDpF