Atmosphere-NX/Atmosphere #2564: Crash when launching game with mods (LayeredFS)

nadrino, 2025-06-17:
## Bug Report

### What's the issue you encountered?

When I updated AtmosphèreNX to the latest 1.9.1, I couldn't launch my game (Star Wars: Knights of the Old Republic II) with mods. Atmosphère crash just after the "Nintendo Switch" logo appears while running the game.

I've originally updated atmosphere to get OFW 20.1.1 running, but this issue still applies for 19.0.1 if running on AtmosphèreNX 1.9.1. This issue was not present for 1.8.0.

I can run the game fine if no mods are applied tho.

### How can the issue be reproduced?

Yes, even after a clean SD install and even a clean NAND restore

### Crash Report

Will be uploaded soon.

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

19.0.1 -- 20.1.1

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusèe
- Official release or unofficial build: 
  - Official release 1.9.1
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: N/A
- EmuMMC or SysNAND:
  - SysNAND

### Additional context?

- Additional info about your environment:
  - Here is a screenshot

[image]

SciresM, 2025-06-17:
Probably this game needs memory stolen in order to build the romfs image, probably huge file count?

nadrino, 2025-06-17:
There are a lot of files, some a big ~100MB yeah. I tried with a reduced version of the mods, a lot less files but it's still the same.

It works perfectly fine in 1.8.0

impeeza, 2025-06-18:
> ## Bug Report
> ### What's the issue you encountered?
> 
> When I updated AtmosphèreNX to the latest 1.9.1, I couldn't launch my game (Star Wars: Knights of the Old Republic II) with mods. Atmosphère crash just after the "Nintendo Switch" logo appears while running the game.
> 
> I've originally updated atmosphere to get OFW 20.1.1 running, but this issue still applies for 19.0.1 if running on AtmosphèreNX 1.9.1. This issue was not present for 1.8.0.
> 
> I can run the game fine if no mods are applied tho.
> ### How can the issue be reproduced?
> 
> Yes, even after a clean SD install and even a clean NAND restore
> ### Crash Report
> 
> [report_0000000030c82622.bin.zip](https://github.com/user-attachments/files/20783445/report_0000000030c82622.bin.zip)
> 
> [01750189802_0100000000000025.log](https://github.com/user-attachments/files/20783461/01750189802_0100000000000025.log)
> ### System Firmware Version
> 
> 19.0.1 -- 20.1.1
> ### Environment?
> 
>     * What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusèe
> 
>     * Official release or unofficial build:
>       
>       * Official release 1.9.1
> 
>     * Do you have additional kips or sysmodules you're loading:
> 
>     * Homebrew software installed: N/A
> 
>     * EmuMMC or SysNAND:
>       
>       * SysNAND
> 
> 
> ### Additional context?
> 
>     * Additional info about your environment:
>       
>       * Here is a screenshot
> 
> 
> [image]

Have you read the Release text?

[image]

nadrino, 2025-06-18:
Thank you @impeeza , didn't see the message! I see @SciresM already responded :) Would there be a way at least to put an explicit error message instead of the crash screen?

chrome99001, 2025-07-11:
So, is this issue fixable?

commit c77acb32 (2025-07-22): fs.mitm: steal some heap when building romfs for kotor2 (closes #2564)
closed 2025-07-22 by SciresM