Atmosphere-NX/Atmosphere #2568: Crash when launching emuMMC

AndreaSecret, 2025-06-23:
## Bug Report

[ * ]</br>

### What's the issue you encountered?
As another user mentioned #2563, i had the same problem while trying to do my periodical update
by updating atmosphere and hekate to their respective latest version.
My main objective was to update the fw to 20.0.0 at least but i couldnt even reach emuMMC to do so

### How can the issue be reproduced?
By launching emuMMC from hekate

### Crash Report

<script src="https://gist.github.com/AndreaSecret/d4cd004d80b459cbbd8143f3d8dbc091.js"></script>

### System Firmware Version

18.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
Hekate
- Do you have additional kips or sysmodules you're loading:
- No
- EmuMMC or SysNAND:
-  EmuMMC that is partition based 

### Additional context?

- Additional info about your environment:
So this crash occured firstly when i tried to open ocarina of time from n64 expansion pack from my emuMMC, so i though that i needed to do some proper upgrading for the corresponding files, but then i could not boot the setup again. Hekate is upgraded to the currently latest version and there seems to be no problem in regards of a corruption of emuMMC (from the info tab in hekate it says that is good to use)

borntohonk, 2025-06-23:
> https://gist.github.com/AndreaSecret/d4cd004d80b459cbbd8143f3d8dbc091
> 
> this should be the correct gist

The crash you've referenced at https://github.com/Atmosphere-NX/Atmosphere/issues/2563 is completely unrelated to your crash report that your logs say.

AndreaSecret, 2025-06-23:
https://gist.github.com/AndreaSecret/f049f95aca0e5718bd93931d27909931