Atmosphere-NX/Atmosphere #2569: Has Atmosphere added USB3.0 support in 20.1.5?

hukaka22, 2025-06-24:
Has Atmosphere added USB3.0 support in 20.1.5?

masagrator, 2025-06-24:
20.1.5 usb sysmodule was not updated, so there is no need.

impeeza, 2025-06-24:
Patches for 20.1.0 & 20.1.1 are the same than 20.1.5 so the released 1.9.1 do not have the patches on it. 

You need to build Commit d8a37b4 to have them on the code and enable the patches.

EXO5KELETON, 2025-06-25:
So how do we get this working on 20.1.5?

Konzolista, 2025-06-25:
> Patches for 20.1.0 & 20.1.1 are the same than 20.1.5 so the released 1.9.1 do not have the patches on it.
> 
> You need to build Commit [d8a37b4](https://github.com/Atmosphere-NX/Atmosphere/commit/d8a37b4b7184b80ba979bcceb98365b8365a1c3a) to have them on the code and enable the patches.

And wouldn't something simple be handy for users who don't know how to code...? ;)

ZweiBR, 2025-06-25:
> Has Atmosphere added USB3.0 support in 20.1.5?

https://github.com/Atmosphere-NX/Atmosphere/issues/2502#issuecomment-2979237053

here

Niemandausduisburg, 2025-07-19:
@impeeza My Nintendo Switch does have USB-Connection issues since latest Atmosphere. Sometimes MTP-Connection isn't working.

closed 2025-07-22 by SciresM

Niemandausduisburg, 2025-10-05:
With FW:20.5.0 the connection issue with USB 3.0 is back.

GoodmanBCN10, 2025-10-05:
> With FW:20.5.0 the connection issue with USB 3.0 is back.
In my case, it "almost always" works fine, except for a few specific games like NBA 2K26, which is 7GB and the update is 53GB.
After a while, the message "Transfer aborted" appears.
I have 20.5.0|1.9.5AMS

ZweiBR, 2025-10-05:
here for now it works well