Atmosphere-NX/Atmosphere #2570: Error Code: 2168-0002 (0x4a8)

Lopeses, 2025-06-24:
Booted into sysCFW on the latest update and about 10 seconds after booting regardless of what is done I get slapped with this error code.

Lopeses, 2025-06-24:
I rewrote my issue so please help.

T3ll3S-05, 2025-06-24:
> I rewrote my issue so please help.
You are using sys-patch aren't you?

T3ll3S-05, 2025-06-24:
> > I rewrote my issue so please help.
You are using sys-patch aren't you?

Lopeses, 2025-06-24:
> no piracy supported.
> 
> delete sys-patch / other custom sysmodules all located in /atmosphere/contents/

Is there any other way? I dont pirate any games, but I can't update Super Smash Bros because I got banned from online by forgetting to keep airplane mode on. So I need to be able to update my game in order to use mods but I can not do it through the EShop.

borntohonk, 2025-06-24:
> Is there any other way? I dont pirate any games, but I can't update Super Smash Bros because I got banned from online by forgetting to keep airplane mode on. So I need to be able to update my game [...]

No.