Atmosphere-NX/Atmosphere #2573: Games not appearing on Sysnand

RottedFish, 2025-07-04:
## Bug Report

Loading my CFW Sysnand results in all titles being grayed out, attempted to load in an endless loop. Games load perfectly fine on StockFW.

Note: there are no pirated games on the system. Though there are translation mods for titles like Higurashi and Umineko, both games are owned legitimately as cartridges.

This issues occurs when loading via Nyx/Hekate, and Tegra Rcm gui to directly load fusee. 

I’ve attempted to reinstall the most recent Atmosphere update, but it didn’t fix the issue.

SciresM, 2025-07-04:
Assuming no piracy, this will be a problem with your custom sysmodules/mods. Delete or rename atmosphere/contents and the issue should go away. All other possible causes are piracy.

SciresM, 2025-07-04:
Grayed out icons in particular are I think sys-tweak being broken on 20+.

closed 2025-07-04 by SciresM