Atmosphere-NX/Atmosphere #2576: 2123-0011 Crash on hOS 20.0.0, PRODINFO not blanked, hosts configured.

enaisnotanumber, 2025-07-11:
## Bug Report

### What's the issue you encountered?

Crash in latest AMS on Horizon OS 20.0.0

Upon connecting to Wi-Fi, error code `2123-0011` is given from program `0100000000000025`. PRODINFO blanking is disabled and hosts are configured to block all Nintendo servers. the console is **entirely** banned ("super ban" i believe is what it is referred to).  

I am using a fresh, just initialized console and fresh Atmosphere installation.  

### How can the issue be reproduced?

- Format SD card
- Add latest Hekate
- Add latest Atmosphere
- Use Daybreak to upgrade from 18.x.x (forgot exact version, not sure if it matters)
- Initialize console
- Setup

In OoBE and not OoBE, it will panic as soon as it connects to Wi-Fi.

### Crash Report

- https://gist.github.com/enaisnotanumber/396c6305e2befe3f6da833b8e0200f38
- https://gist.github.com/enaisnotanumber/5f84bcc4cce3ecee5194deabcd009a16

### System Firmware Version

20.0.0

### Environment?

- Latest Hekate
- Latest Atmosphere
- Stock HorizonOS, no games.

---
this might be a possible dupe but i cannot find anything else not referencing PRODINFO

borntohonk, 2025-07-11:
> the console is **entirely** banned ("super ban" i believe is what it is referred to).
> In OoBE and not OoBE, it will panic as soon as it connects to Wi-Fi.
> hosts are configured to block all Nintendo servers. 

> https://gist.github.com/enaisnotanumber/396c6305e2befe3f6da833b8e0200f38

Right so this is symptomatic, or rather, identical (literally +/- a few bytes offset) to the crash condition set forth by failing to initiate https communication against the "time" domain. 

obtain a patch for "NIM" to rectify this, or properly use dns.mitm (i know your post states you are using it, but your symptom indicates you aren't)

> https://gist.github.com/enaisnotanumber/5f84bcc4cce3ecee5194deabcd009a16

this one afaik indicates you've aborted a firmware update unintentionally

https://suchmememanyskill.github.io/guides/unbrick/#early-boot-errors
https://suchmememanyskill.github.io/guides/unbrick/#mmc-rebuild

enaisnotanumber, 2025-07-12:
thank you, i will close the issue

closed 2025-07-12 by enaisnotanumber