Atmosphere-NX/Atmosphere #2585: 20.2.0 support

Brkmnly, 2025-07-15:
nintendo tricked me into updating thx nintendo love ya

ELY3M, 2025-07-15:
the latest commit worked for me on 20.2

ChriF1514256, 2025-07-15:
The atmosphere go on the 20.2.0 ??

InfinityCoda, 2025-07-15:
SciresM's already said that he'll post a release for 20.2.0 support tonight or tomorrow morning, just give him time

ChriF1514256, 2025-07-15:
Ok

ChriF1514256, 2025-07-15:
Thx alot

scrapp08, 2025-07-15:
Every time I try boot into emummc it just goes blank, I’m assuming this is the expected behaviour until the fixed 20.2.0 release comes out?

closed 2025-07-15 by SciresM

ChriF1514256, 2025-07-15:
I think yes

ChriF1514256, 2025-07-16:
@SciresM we communities believe in you and we can't wait to see the cool update that you have made anxiously look forward to the file and above all I hope one day to be able to interact and help even developing skills like yours for me you are very inspiring thank you very much

zendy199x, 2025-07-16:
I also encountered the problem when updating to version 20.2.0 with the error "a fatal error occurred when running atmosphere 010041544d530000" even though I tried everything but could not fix it, I found out that Atmosphere does not support it yet, please try to release as soon as possible to support 20.2.x

SonntagGit, 2025-07-16:
I’m new here. Where can I find the Atmosphère version that supports 20.2.0? So far, I only find the latest release, Atmosphère 1.9.1.

dxgldotorg, 2025-07-17:
> We are waiting for new release

It's here.  https://github.com/Atmosphere-NX/Atmosphere/releases/tag/1.9.2