Atmosphere-NX/Atmosphere #2586: Atmosphere support for 20.2.0

FLIPPERZERO1200, 2025-07-16:
I accidentally updated my switch to 20.2.0 and when I tried to boot it, it showed the purple boot screen and then it showed a black screen. I am trying to downgrade for now but can you guys pls make basic support for the new HOS? Thanks! (https://www.reddit.com/r/SwitchPirates/s/W8pSdH3dqG)

InfinityCoda, 2025-07-16:
Duplicate of #2585 and others, an update is already in the words, just wait patiently. The link to the piracy community is completely unnecessary. From SciresM earlier:

> I made the relevant code changes from a laptop last night, I just don't have access to a PC to do a build/release until tonight or tomorrow morning.

SonntagGit, 2025-07-16:
I’m new here. Where can I find the Atmosphère version that supports 20.2.0? So far, I only find the latest release, Atmosphère 1.9.1.

zendy199x, 2025-07-16:
We are waiting for new release

UPGRADE-VINTED, 2025-07-16:
I have this. It doesn't have the official update, but it works on 20.2 (it's an edit of version 1.19.1).
https://www.swisstransfer.com/d/b55ed765-9273-4d1b-a00f-c6f4ca8f2e7e

I still recommend not updating and, above all, waiting for the official Atmosphere version (but for those like me who didn't want to wait).

closed 2025-07-16 by SciresM