Atmosphere-NX/Atmosphere #2588: switch is unable to resume from screen turning off.

likehelly, 2025-07-18:
## Bug Report

### What's the issue you encountered?

updated to latest atmosphere, and hekate, releases. now when my switch's screen turns off, it will not come back until a hard reset is done. 

### How can the issue be reproduced?

enter rcm, push payload, boot into the os, let the screen turn off and try to turn it back on. 

### Crash Report

there is no crash report.

### System Firmware Version

20.1.5

T3ll3S-05, 2025-07-19:
I don't have such issue and I've tried both 20.1.5 and 20.2.0, maybe something is wrong with the SD card or the switch itself?

Bulboussum, 2025-07-19:
I have the same issue with no crash reports but on system version 20.1.1.

vas2525, 2025-07-19:
I don't have that issue. on any 20.0.0 - 20.2.0 firmwares

likehelly, 2025-07-20:
in an attempt to fix things, and just because i wanted to switch to a larger sd card, i have redid everything fresh, no files have been transferred between sd cards, and both are legitimate sandisk sd cards bought at walmart so unlikely to be fake, but anyways.

it still does it.

Bulboussum, 2025-07-20:
Downgrading Atmosphere to 1.9.1 fixed the issue for me.

closed 2025-07-20 by SciresM