Atmosphere-NX/Atmosphere #2610: TOTK Atmosphere Error

MaikonRavenhurst, 2025-08-26:
When I try to install any modification and launch the game, the Atmosphere fatal error crashes. I tried to install mods via TKMM-nx, as well as manually.
Atmosphere: 1.9.3
heKate: 6.3.1
Base: Ultra-NX
TOTK: 1.2.1 (Also tried on 1.4.2).
What's problem?
Attachment: Atmosphere error file.
[report_000000001958e31c.zip](https://github.com/user-attachments/files/21983669/report_000000001958e31c.zip)

MaikonRavenhurst, 2025-08-26:
> > TKMM-nx
> > Base: Ultra-NX
> 
> https://github.com/TKMM-Team/TKMM-NX/issues https://github.com/Ultra-NX/UltraNX/issues
> 
> seek support from your bundles, not here.

What does TKMM have to do with it, even if Atmosphere crashing at game startup with manual installation of mods?

borntohonk, 2025-08-26:
> > > TKMM-nx
> > > Base: Ultra-NX
> > 
> > 
> > https://github.com/TKMM-Team/TKMM-NX/issues https://github.com/Ultra-NX/UltraNX/issues
> > seek support from your bundles, not here.
> 
> What does TKMM have to do with it, even if Atmosphere crashing at game startup with manual installation of mods? And about Ultra-NX - I changed to Kefir. Same problem with Atmosphere.

https://github.com/rashevskyv/kefir/issues

seek support in the correct places. You are not using atmosphere.

You are using modified/bundles of atmosphere.

remove all elements not atmosphere, if problem persist, submit proper issue without deleting the issue template.

closed 2025-08-26 by MaikonRavenhurst