Atmosphere-NX/Atmosphere #2611: [Feature Request] Graceful shutdown on low power

HeyItsJono, 2025-08-29:
## Feature Request

### What feature are you suggesting?
#### Overview:
- When the Switch is booted in to CFW on EmuMMC and is allowed to completely discharge its battery, it frequently leads to EmuMMC corruption and failure to boot into EmuMMC when next charged (routinely there is no error, just a plain black screen on attempting to next boot into emuMMC atmosphere)
- A feature to formally power down the console when the battery is depleted to a user-set threshold (e.g. <5%) would prevent such issues from occurring in the future

#### Smaller Details: 
N/A

#### Nature of Request:  
[ Remove all that do not apply to your request. ]
- Addition
  - Optional feature that allows a user to set a battery percentage threshold under which Atmosphere will automatically gracefully power down the console
  - Ideally, a pop-up that warns the user when this is about to take place, with a short grace period timer that allows the user to cancel the shutdown if they're actively using the console and go plug it in
  - Alternatively, make the auto-shutdown only occur if the console is in sleep mode

### What component do you feel this would best fit within?
Any of:
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
    - Custom system modules.</br>
- [**Thermosphère**](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Atmosphère's emuNAND implementation.</br>
- [**Troposphère**](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Application-level patches to the Horizon OS.</br>

[ Note: **Bolded components are not implemented** or are still at the prototyping phase. ]

### Why would this feature be useful?
Over the last few years I've personally had my emuMMC corrupted multiple times by this issue, leading to days of trying to recover saves and games from the corrupted emuMMC and porting them over to the new one. I try to avoid it best as I can by being mindful to either leave my switch on charge or power it down myself when not in use, but the reality is life happens and occasionally being human I slip up and am once again met with this issue. I'm not the only one either, there are countless posts online of others having this issue.

Implementing an auto-shutdown process would avoid this issue. Keeping it as an optional flag would mean for those who don't think it is useful or would find it a hassle, it can just be turned off.

SciresM, 2025-08-29:
They are probably using an exFAT formatted sd card instead of FAT32. This causes corruption on non-graceful shutdown pretty regularly.

HeyItsJono, 2025-09-01:
> They are probably using an exFAT formatted sd card instead of FAT32. This causes corruption on non-graceful shutdown pretty regularly.

I've had my Switch hacked since the early days of public releases in the scene, and not once have ever had my SD card formatted as exFAT; pretty much every guide out there warns you to only use FAT32. Despite that, I've had this happen at least 11 or 12 times over the last 3-4yrs. It has happened across multiple different SDs, all from reputable brands purchased from brick and mortar stores, all of which had been tested at time of purchase and at time of corruption with read/write/verify tests, all of which were error-free, and all of which were formatted as FAT32. My brother has experienced this issue on his own switch multiple times, also with a FAT32 SD, and I've had a couple of mates have the same thing happen to them. It might not be a common issue, it might not be easy to recreate, it certainly doesn't happen every single time the battery dies, but it definitely exists.